innoveit/react-native-ble-manager #7: Scan not filtering on Android ?

NabilEL, 2016-06-20:
Scanning with a service UUID doesn't seem to filter anything, I still discover every peripheral around.
But it's working on iOS.

Tested on HTC M7, Samsung GT-i9505 (both running 5.0.1) and iPhone 6 Plus running iOS 9

marcosinigaglia, 2016-06-20:
Hi, it's an Android bug, the Android module now use only the 18 API, it's necessary to add the support of new API.

NabilEL, 2016-06-20:
Alright, thanks for the answer.
I went ahead and un-commented the line with the services filter in the android code and it's working.

marcosinigaglia, 2016-06-20:
Can you try to fix the code starting from this?

```javascript
	@ReactMethod
	public void scan(ReadableArray serviceUUIDs, final int scanSeconds, Callback successCallback) {
		Log.d(LOG_TAG, "scan");
		if (!getBluetoothAdapter().isEnabled())
			return;

		if (Build.VERSION.SDK_INT >= LOLLIPOP) {
			BluetoothLeScanner scanner = getBluetoothAdapter().getBluetoothLeScanner();

			ScanSettings settings = new ScanSettings.Builder().setScanMode(ScanSettings.SCAN_MODE_LOW_LATENCY).build();
			ArrayList<ScanFilter> filters = new ArrayList<ScanFilter>();
			scanner.startScan(filters, settings, mScanCallback);

			if (scanSeconds > 0) {
				Thread thread = new Thread() {

					@Override
					public void run() {

						try {
							Thread.sleep(scanSeconds * 1000);
						} catch (InterruptedException e) {
						}

						runOnUiThread(new Runnable() {
							@Override
							public void run() {
								BluetoothLeScanner scanner = getBluetoothAdapter().getBluetoothLeScanner();
								scanner.stopScan(mScanCallback);
								WritableMap map = Arguments.createMap();
								sendEvent("BleManagerStopScan", map);
							}
						});

					}

				};
				thread.start();
			}

		} else {
			for (Iterator<Map.Entry<String, Peripheral>> iterator = peripherals.entrySet().iterator(); iterator.hasNext(); ) {
				Map.Entry<String, Peripheral> entry = iterator.next();
				if (!entry.getValue().isConnected()) {
					iterator.remove();
				}
			}

			if (serviceUUIDs.size() > 0) {
				UUID[] services = new UUID[serviceUUIDs.size()];
				for(int i = 0; i < serviceUUIDs.size(); i++){
					services[i] = UUIDHelper.uuidFromString(serviceUUIDs.getString(i));
					Log.d(LOG_TAG, "scan su service: " + serviceUUIDs.getString(i));
				}

				//getBluetoothAdapter().startLeScan(services, mLeScanCallback);
				getBluetoothAdapter().startLeScan(mLeScanCallback);
			} else {
				getBluetoothAdapter().startLeScan(mLeScanCallback);
			}

			if (scanSeconds > 0) {
				Thread thread = new Thread() {

					@Override
					public void run() {

						try {
							Thread.sleep(scanSeconds * 1000);
						} catch (InterruptedException e) {
						}

						runOnUiThread(new Runnable() {
							@Override
							public void run() {
								getBluetoothAdapter().stopLeScan(mLeScanCallback);
								WritableMap map = Arguments.createMap();
								sendEvent("BleManagerStopScan", map);
							}
						});

					}

				};
				thread.start();
			}
		}
		successCallback.invoke();
	}

	private ScanCallback mScanCallback = new ScanCallback() {
		@Override
		public void onScanResult(int callbackType, ScanResult result) {
			Log.i("callbackType", String.valueOf(callbackType));
			Log.i("result", result.toString());
			BluetoothDevice btDevice = result.getDevice();
			Log.i(LOG_TAG, "DiscoverPeripheral: " + btDevice.getName());
			String address = btDevice.getAddress();

			if (!peripherals.containsKey(address)) {

				Peripheral peripheral = new Peripheral(btDevice, result.getRssi(), result.getScanRecord().getBytes(), reactContext);
				peripherals.put(btDevice.getAddress(), peripheral);

				BundleJSONConverter bjc = new BundleJSONConverter();
				try {
					Bundle bundle = bjc.convertToBundle(peripheral.asJSONObject());
					WritableMap map = Arguments.fromBundle(bundle);
					sendEvent("BleManagerDiscoverPeripheral", map);
				} catch (JSONException e) {

				}

			} else {
				// this isn't necessary
				Peripheral peripheral = peripherals.get(address);
				peripheral.updateRssi(result.getRssi());
			}
		}

		@Override
		public void onBatchScanResults(List<ScanResult> results) {
			for (ScanResult sr : results) {
				Log.i("ScanResult - Results", sr.toString());
			}
		}

		@Override
		public void onScanFailed(int errorCode) {
			Log.e("Scan Failed", "Error Code: " + errorCode);
		}
	};
```

marcosinigaglia, 2016-06-21:
I add a check on Android version for the filter.

closed 2016-06-21 by marcosinigaglia