innoveit/react-native-ble-manager #9: "undefined" to Attempts to connect to a peripheral

vlade2008, 2016-06-21:
[image]
this is my code, please help me. thanks

marcosinigaglia, 2016-06-21:
Hi, before connect you have to wait the scan has found the peripheral.

NabilEL, 2016-06-21:
You have to subscribe to the BleManaDiscoverPeripheral event, and inside it you will be able to connect to the device you want (you still have to scan though) : 

Example : 
```
      BleManager.scan([tabletAppServiceUUID],60)
            .then(() => {
                console.log('scan started');
            })
            .catch((error) => {
                console.log("error scan : " + error);
            });
       NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', (args) => {
           // Connect to your device here
       }
```

closed 2016-06-21 by marcosinigaglia

tarang9211, 2016-07-22:
I tried the above code, and I still cannot connect to my MAC from my iPhone. This is the code

 ```
 btnPress() {
    BleManager.scan([], 60)
    .then(() => {
        console.log('scan started');
    });

    DeviceEventEmitter.addListener('BleManagerDiscoverPeripheral', (args) => {
        console.log(args);
        BleManager.connect('F71DA437-87A4-5E34-BC08-836D06B2FEAF')
        .then(() => {  
            console.log('success')
        })
        .catch((error) => {
            console.log(error);
        });
    })
  }
```