innoveit/react-native-ble-manager #11: NativeAppEventEmitter  is wrong

ysjiang4869, 2016-07-08:
AwersomeProjeect, few questions,
one is NativeAppEventEmitter should be DeviceEventEmitter according to the docs of react-native;
Another question when i connect to an peripheral it will disconnect after a few minutes automatically; I don't know the reason , can you give me some help or hint? thanks

marcosinigaglia, 2016-07-08:
Hi @ysjiang4869 , in the docs is not specified the difference between the two emitters so i choose the NativeAppEventEmitter with no reasoning. Why do you think the DeviceEventEmitter is better?

ysjiang4869, 2016-07-11:
@marcosinigaglia   I  find NativeAppEventEmitter in earlier docs but now is DeviceEventEmitter, when i use NativeAppEventEmitter  it tells me not found,so i think may be the first one was deprecated

marcosinigaglia, 2016-07-11:
It's not deprecated, maybe I can move to DeviceEventEmitter because the module has device related events.

closed 2016-07-11 by marcosinigaglia