innoveit/react-native-ble-manager #12: Cannot read property 'scan' of undefined

morioniot, 2016-07-14:
Hi, 

Thanks for this excellent job. I was trying to scan for devices with this module but I can not make it work. I am developing an Android app using a Samsung Galaxy S3 phone with Android 4.3. I am also using the last version of React Native (0.29). I did all the configurations described in the README file, but when I try to run the application appears the error: "cannot read property 'scan' of undefined". Could you please help me with this problem?
[image]

marcosinigaglia, 2016-07-15:
Hi, are you sure you add the BleManagerPackage in MainApplication?

morioniot, 2016-07-15:
Hi Marco,
I am sure I did that. I modify the file android/app/src/main/java/com/{nameOfMyProject}/MainApplication.java Here is a screenshot:
[image]

marcosinigaglia, 2016-07-15:
Have you try to clean the node_modules dir and install the modules again?

morioniot, 2016-07-15:
I cleaned the node_modules and installed them again. The error did not disappear.

marcosinigaglia, 2016-07-15:
Is null the BleManager variable?

morioniot, 2016-07-15:
Hi,
I think that the problem is in the root file of the module. The variable bleManager is undefined. So, maybe there is a problem in this line:

`var bleManager = React.NativeModules.BleManager;`

Here is the screenshot of the developer tools:
[image]

marcosinigaglia, 2016-07-15:
Check in the device logs if it appears "BleManager initialized"

morioniot, 2016-07-15:
Hi Marco,
I found what was the cause of the error. First, I had not re-compiled the entire project with "react-native run-android". When I did that I also found that i had to change the minSdkVersion in the build.gradle file. Now, everything is working just fine. I apologize if I represented a waste of time. Thanks a lot for all your help.

closed 2016-07-17 by marcosinigaglia