innoveit/react-native-ble-manager #15: Bluetooth connection examples

tarang9211, 2016-07-19:
Could you post some more examples on how to connect to a device?

I tried using the examples from the README.md to connect my iphone to my mac using the UUID and MAC address, but none of them worked.

Thanks!

marcosinigaglia, 2016-07-20:
Hi, you have to scan for the peripherals and then use the id of the peripheral to connect.

closed 2016-07-20 by marcosinigaglia

tarang9211, 2016-07-21:
So first I would call `BleManage.scan()` and then `connect` right? So will the `scan()` method return a list of UUID or MAC addresses of active bluetooth devices?

marcosinigaglia, 2016-07-21:
First you have to handle BleManagerDiscoverPeripheral event, then start the scan and you receive the peripheral finded.

tarang9211, 2016-07-22:
Hi Marco, I tried that. Does this code look right?

`DeviceEventEmitter.addListener('BleManagerDiscoverPeripheral', (args) => {
        console.log(args);
    });

    BleManager.scan([], 5, true)
    .then(() => {
        console.log('here');
    })
`

marcosinigaglia, 2016-07-25:
Try this:

```javascript
componentDidMount() {
  this.handleDiscoverPeripheral = this.handleDiscoverPeripheral.bind(this);

  NativeAppEventEmitter.addListener(
    'BleManagerDiscoverPeripheral',
    this.handleDiscoverPeripheral
  ); 
}

btnPress() {
  BleManager.scan([], 5);
}

handleDiscoverPeripheral(data){
  console.log(data);
}
```

DarrylD, 2016-08-26:
That example would be awesome for documentation example