innoveit/react-native-ble-manager #18: How to read output from bluetooth device?

austinksmith, 2016-07-28:
It seems this is (and all the other similar projects I've come across) are simply bluetooth managers IE. you can disconnect and connect to devices from within in app which seems rather odd since the OS can typically handle that on its own. Perhaps I'm not understanding the documentation correctly but how can I do something like this..

```
bluetoothHeartMonitor.listenForInput(input) {
 this.setState{
    pulse: input;
  }
}

render() {
  <text>{this.state.pulse}</text>
}
```

Where input is the value that came from the bluetooth device. Any help is much appreciated, thanks.

marcosinigaglia, 2016-07-29:
Hi, read this [documentation](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.heart_rate.xml&u=org.bluetooth.service.heart_rate.xml). I think you have to connect to the device and start notification on service `180D` and characteristic `2A37`.

closed 2016-08-01 by marcosinigaglia

austinksmith, 2016-08-01:
Thanks, I will follow up if I have any more concerns. The heartbeat monitor was just an example, the device i'll be interacting with is going to be a barcode scanner.