innoveit/react-native-ble-manager #21: Device name

vincenzoiacovone, 2016-08-05:
Why device name is always null? I tried with iOS

Bluetooth.scan([], 5, false)
    .then(() => {
      // Success code
      console.log('Scan started');
    });

NativeAppEventEmitter.addListener(
      'BleManagerDiscoverPeripheral',
      (args) => {
          console.log(args)
      }
    );

args is object like this:
{ id: 'C92C740D-A084-FAB7-7A44-251D12F7D099',
  advertising: { kCBAdvDataIsConnectable: true } }
2016-08-05 18:43:12.399 [info][tid:com.facebook.react.JavaScript] { id: 'C92C740D-A084-FAB7-7A44-251D12F7D099',
  advertising: 
   { kCBAdvDataServiceData: 
      { '8EB5': 
         { CDVType: 'ArrayBuffer',
           data: 'AZAGmKhr6ZF2x0/EiSpVGs8Hx3wADgK7+ANY' } },
     kCBAdvDataIsConnectable: true } }

marcosinigaglia, 2016-08-05:
Hi, maybe the BLE device has no name.

vincenzoiacovone, 2016-08-05:
The name exists, if i search devices in bluetooth section, i found it with his name..

marcosinigaglia, 2016-08-08:
Do you see any error on XCode log?

vincenzoiacovone, 2016-08-08:
No errors. It is possible that my device is not low energy or the version is Lower then 4.0? I read that corebluetooth framework is working only with low energy and an uuid is required for scan.

http://stackoverflow.com/questions/10178293/how-to-get-list-of-available-bluetooth-devices

marcosinigaglia, 2016-08-08:
Try an app like SmarDiscover or LightBlue and see the scan results.

vincenzoiacovone, 2016-08-08:
No results with this apps too. It supported only ble peripherals?

marcosinigaglia, 2016-08-08:
On iOs you have limitation by [MFI](https://mfi.apple.com/MFiWeb/getFAQ.action).

closed 2016-08-08 by marcosinigaglia

RameshAmzur, 2019-05-24:
Hi 
In android also getting name as null. Tested with another application and it is showing device names.

EduardoRicardez, 2020-05-21:
In android also getting name as null too