innoveit/react-native-ble-manager #23: Why I should convert data to base64 format?

vdzundza, 2016-08-08:
Hi there,
I've investigated a source code of cordova plugin. I see that I can just send a array of bytes
https://github.com/don/cordova-plugin-ble-central/blob/15a8c204779357c2e3c3bb6370987ea6ff38b92a/src/android/Peripheral.java#L509

So what is purpose for converting string to base64 for?

marcosinigaglia, 2016-08-09:
Hi, I tried to pass directly the array but I had some problems so the simple way to fix the problem is to convert to a string, maybe I can convert directly in the module.

closed 2016-08-10 by marcosinigaglia

ibenben, 2017-05-26:
I want to send a array of bytes.How can I do this? @marcosinigaglia @furydam

marcosinigaglia, 2017-05-26:
```js
var base64 = require('base64-js');
var data = base64.fromByteArray(yourData);
```

ibenben, 2017-05-27:
@marcosinigaglia    Thanks.