innoveit/react-native-ble-manager #24: Can't read advertisment raw data.

henninghall, 2016-08-11:
Hey! I am trying to decode the raw data of the advertising package (advertising.data). First I decoded the string to a byte array with bas64-js package and then I tried the following node packages:

https://github.com/tessel/bleadvertise - Didn't work. "Can't find variable: Buffer"
https://www.npmjs.com/package/ble-ad-parser - - Didn't work. "Can't find variable: Buffer"
https://github.com/mfgCode/ble-packet-parser - Didn't work. Reports are undefined. 

Does anyone know how to read the data? :) 

Example how the whole package looks like. I am trying to decode the advertising.data part.  
```
{
rssi: -93,
name: 'Liee_A3',
id: '4E:3C:B1:F7:A0:F9',
advertising: 
{ CDVType: 'ArrayBuffer',
data: 'AgEGE/9MAAwOAEGAkOLfaaB9WE7cR0AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=' } 
}
```

mzsolt, 2016-08-11:
Hi!

Node core modules aren't supported in React Native. You should try [react-native-browser-builtins](https://github.com/originalgremlin/react-native-browser-builtins) or [buffer](https://github.com/feross/buffer) to fix `bleadvertise` and `ble-ad-parser` packages.

I used this package to monitor kontakt.io iBeacons. These beacons have a D00D service data field containing the unique id, firmware version, and battery level.
To extract the battery level from the last byte, I converted the data to string with `atob` and took the last character's value with `.codePointAt()` (ES6).

Hope this helps.

closed 2016-09-02 by marcosinigaglia

MehdiZaghouani, 2018-03-02:
Hi, please how do you solve this problem ?? and thank you

savyad, 2018-08-22:
hey how can i read advertisement data from the devices