innoveit/react-native-ble-manager #26: hi i'm keeping getting error ' The callback write() exists in module BleManager, but only one callback may be registered to a function in a native module.'

mnkang89, 2016-08-17:
I want to scan some my bluetooth module and connect to it. after i connected to it, i want to write data to specific characteristic UUID. but i've kept getting error 

>  The callback write() exists in module BleManager, but only one callback may be registered to a function in a native module.

my whole code is below

import React, { Component } from 'react';
import {
  AppRegistry,
  StyleSheet,
  Text,
  View,
  TextInput,
  Image,
  ActivityIndicator,
  TouchableHighlight,
  NativeAppEventEmitter 
} from 'react-native';

var Button = require('../common/button');
var BleManager=require('react-native-ble-manager');
var base64 = require('base64-js');

var b_list;
var dev;
var sub;

module.exports = React.createClass({
  getInitialState: function(){
    return {
      username: '',
      password: '',
      animating: true,
      service: 'ddsjdsljslks',
    };
  },

  render: function(){
    return (
      <View style={ styles.container2 }>
        <View style={ styles.p1 }>
          <Text style={styles.t1}>Sign Out</Text>
          <TouchableHighlight onPress = { this.btnPress }>
            <Text>Bluetooth test</Text>
          </TouchableHighlight>
          <Text>{this.state.service}</Text>

          <TouchableHighlight
            style={styles.button}
            underlayColor={'gray'}
            onPress={this.onSkipPress}
            >
            <Text style={styles.t3}>Skip</Text>
          </TouchableHighlight>

        </View>

        <View style={styles.p2}>
          <Image
            style={styles.img1}
            source={{uri: 'https://d30y9cdsu7xlg0.cloudfront.net/png/37728-200.png'}
          }/>

          <View style={styles.p2_row}>
            <ActivityIndicator animating={this.state.animating} color="black" style={ {paddingRight:15} }/>
            <Text>Your device has Blueooth {"\n"} turned off. Please swipe up {"\n"} and turn it on.</Text>
          </View>
        </View>

        <View style={styles.p3}>
          <Text>Need help with pairing?</Text>
        </View>

      </View>
    );
  },

  componentDidMount: function() {
    this.handleDiscoverPeripheral = this.handleDiscoverPeripheral.bind(this);

    sub= NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral); 

  },

  onSkipPress: function(){
    //this.props.navigator.push({name: 'v2_calm'});
    
    var data = base64.fromByteArray('HeckYes');
    BleManager.write(dev.id, "00002220-0000-1000-8000-00805F9B34FB", "00002221-0000-1000-8000-00805F9B34FB", data)
  },

  onPress: function(){
    // Log the user in
    this.setState({
      password: ''
    });

  },

  btnPress: function(){
    BleManager.scan([], 5);
    
  },

  handleDiscoverPeripheral: function(device){
    dev=device;
    console.log(typeof device);
    console.log(Object.getOwnPropertyNames(device));
    console.log(device);
    if(typeof device===undefined)
      { this.setState({service: 'unde'}) }
    else
      { 
        this.setState({service: device.name});
        BleManager.connect(device.id)
          .then(() => {
            // Success code
            console.log('Connected');
          })
          .catch((error) => {
            // Failure code
            console.log(error);
          });
        //this.props.navigator.push({name: 'v2_calm'});
        sub.remove();
      }
  }

});

Rmannn, 2016-08-19:
sub.remove() can't work as sub is not in the same scope (scope of componentDidMount)
You must store sub in your object itself to keep a reference using `this.sub = ...`
```js
//sotre the sub in this
componentDidMount(){
    this.sub = NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral); 
},
//remove the sub on unmount or in your method
componentWillUnmount(){
    this.sub.remove();
},
```

mnkang89, 2016-08-23:
Thanks! -----Original Message-----

closed 2016-08-23 by marcosinigaglia