innoveit/react-native-ble-manager #45: The connect success callback is never called if the peripheral has no service

haibinyu, 2016-09-06:
I encounter a issue, when my ble device has no service, the connect success callback is never called after I call:

```
BLEManager.connect(peripheralId)
.then(() => {
  // the code here is never called
})
```
After research the code, I find that after connected the peripheral, connect callback is not called, but **discoverServices** method and **discoverCharacteristics:forService:** is called immediately.
I think this is not a best practice and it is better to split  **discoverServices** from **connect**.

marcosinigaglia, 2016-09-06:
Hi, I thinks there is a reason for call the callback after discoverServices, I'll investigate

marcosinigaglia, 2016-09-08:
Hi, temporarily you can use BleManagerConnectPeripheral event (https://github.com/innoveit/react-native-ble-manager/commit/e17ce837d51537a6288895f0f381a013421a3050)

haibinyu, 2016-09-08:
@marcosinigaglia thanks, 🎉

closed 2016-09-10 by haibinyu