innoveit/react-native-ble-manager #49: Is there anyway to make this work on Background even after the app is killed?

sudharsan1988, 2016-09-07:
Hi, I am wondering is there anyway to make this library scans ble data on the Background even after the app is killed. I can see, it works only on the foreground for now. Thanks in advance!

marcosinigaglia, 2016-09-08:
Hi, you need specific native code to handle the ble in background so is not possible to have a generic library.

closed 2016-09-08 by marcosinigaglia

anton-aleksandrov, 2017-03-06:
@marcosinigaglia do you have an example? I understand, generally, how to implement this with `NativeModules` but there are quite a few details i need to discover yet.

hidayatNgganteng, 2018-06-01:
You must add **UIBackgroundModes key** in info.plist and the value is array containing string _bluetooth-central_ and _bluetooth-peripheral_

allowing bluetooth behavior even in the background