innoveit/react-native-ble-manager #54: getConnectedPeripherals return empty array if the serviceUUIDs params is []

haibinyu, 2016-09-10:
when  I call getConnectedPeripherals with serviceUUIDs = [], the result is empty, but if I call it with serviceUUIDs = ['one service uuid in connected peripherals], the result is right.
is the serviceUUIDs required to getConnectedPeripherals?

marcosinigaglia, 2016-09-12:
Hi, in Android or iOS?

haibinyu, 2016-09-12:
@marcosinigaglia iOS

marcosinigaglia, 2016-09-12:
Try version 1.4.2

kevyu, 2016-09-12:
@thanks, I will try tommrow.

closed 2016-09-21 by marcosinigaglia

Abhinandan-Kushwaha, 2021-05-31:
Still facing the same issue

longnguyenITG, 2021-06-30:
i have same issue, but use getBondedPeripherals get success list connected peripherls but have'nt uuids, what different between getBondedPeripherals and getConnectedPeripherals?

SuhitAgarwal, 2024-07-12:
This still seems to be an issue @marcosinigaglia. It was working fine until a few days ago. I can't seem to find issues anywhere else in my code. i'm just not getting any peripherals when I run the `BleManager.getConnectedPeripherals([])`. 

Here's my console.log from running that: `{"_h": 0, "_i": 0, "_j": null, "_k": null}`