innoveit/react-native-ble-manager #55: Example of startNotification / stopNotification?

DblBee, 2016-09-15:
I am unable to get startNotification / stopNotification working. My peripheral is a RPI3 running Node JS Bleno. The characteristic works with the subscribe, read, and write operations using the nRF Connect application on Android. I can get the read and write operations to work calling them individually. All operations fail after the application calls startNotification. Does anyone have a good example of startNotification working?

marcosinigaglia, 2016-09-19:
Hi, can you paste your code?

DblBee, 2016-09-19:
`  BleManager.connect(peripheralId)
    // Uncomment the then() below to add the startNotification
    // .then(() => {
    //   console.log('connected');
    //   return BleManager.startNotification(peripheralId, serviceUUID, characteristicUUID);
    // })
    .then(() => {
      console.log('Notification started');
      return BleManager.write(peripheralId, serviceUUID, characteristicUUID, data);
    })
    .then(() => {
      const resultData1 = new Buffer(data, 'hex');
      console.log(`Write complete: ${resultData1.toString('utf8')}`);
    })
    .then(() => {
      console.log('Attempting to read');
      return BleManager.read(peripheralId, serviceUUID, characteristicUUID);
    })
    .then((readData) => {
      const resultData2 = new Buffer(readData, 'hex');
      console.log(`Read: ${resultData2.toString('utf8')}`);
    })
    .catch((error) => console.log(`error: ${error}`));`

marcosinigaglia, 2016-09-19:
If you use notification I think you have to handle the `BleManagerDidUpdateValueForCharacteristic` event and not read the characteristic.

closed 2016-09-21 by marcosinigaglia