innoveit/react-native-ble-manager #56: problem in introduction

ysjiang4869, 2016-09-19:
use as follow when use in android
protected List<ReactPackage> getPackages() {
        return Arrays.<ReactPackage>asList(
            new MainReactPackage(),
                new BleManagerPackage(this)
        );
    }

marcosinigaglia, 2016-09-19:
Hi, witch version of RN do you use?

ysjiang4869, 2016-09-19:
0.28.0， but in your  BleManagerPackage class the constructor need params Activity activityContext

marcosinigaglia, 2016-09-19:
Sorry but the library dosn't support RN 0.28

closed 2016-09-19 by marcosinigaglia

ysjiang4869, 2016-09-19:
which version is supported? i use 0.27 before and it can work well

marcosinigaglia, 2016-09-19:
You can use [v.0.0.14](https://github.com/innoveit/react-native-ble-manager/releases/tag/v0.0.14), after this version i think works only with RN 0.29+

ysjiang4869, 2016-09-19:
ok,thanks, i still use v0.0.14 so there is no wrong