innoveit/react-native-ble-manager #57: Example lacking instructions

Cacowned, 2016-09-19:
Hi,

The example in the readme doesn't show how to import/require the ble-manager.

I tried

`var BleManager = require('react-native-ble-manager');`

and

`import { BleManager } from 'react-native-ble-manager';`

but both result in an undefined BleManager.

marcosinigaglia, 2016-09-19:
Hi, there is the **Install** section in [README](https://github.com/innoveit/react-native-ble-manager/blob/master/README.md)

closed 2016-09-19 by marcosinigaglia

Cacowned, 2016-09-19:
@marcosinigaglia Yes, there is. But it only tells me how include it to compile with the android or iOS native stuff, but not how to use it on the react-native side, the javascript part.

marcosinigaglia, 2016-09-19:
Try
`import BleManager from 'react-native-ble-manager';`

Cacowned, 2016-09-19:
@marcosinigaglia Ok, that works, thanks!