innoveit/react-native-ble-manager #59: There are no characteristic data after connecting a BLE device

ttdat89, 2016-09-27:
After discovering/connecting a peripheral, there is no characteristic in returned object.

marcosinigaglia, 2016-09-27:
Hi, can you give more information?

ttdat89, 2016-09-28:
Hi, actually i want to ask how to get characteristic UUID.
I saw that you have connectCallbacks that return characteristic, but i cannot get returned data somehow

marcosinigaglia, 2016-10-06:
I thinks the device has no services, the connect callback is called after the services discovery. You have to use BleManagerConnectPeripheral event.

BigPun86, 2016-10-11:
+1 No idea where to obtain the characteristicUUID

marcosinigaglia, 2016-10-12:
Hi, in release 2.2.0 you have the peripherals services and characteristics in connect callback. I will change the docs soon.

closed 2016-10-12 by marcosinigaglia

BigPun86, 2016-10-13:
Tried it, but unfortunately it doesn´t work .... Tried on real device of course :P

[image]

marcosinigaglia, 2016-10-13:
Try 2.2.1

BigPun86, 2016-10-14:
Almost :P This is my crash report ...

[image]
[image]
[image]

marcosinigaglia, 2016-10-14:
Fix in 2.2.3, sorry.

BigPun86, 2016-10-17:
Nice. It is getting better :D But i get now the error: Invalid UUID: 1800. I think the native java part can´t handle this kind of string?!

marcosinigaglia, 2016-10-17:
The services is a UUID, i never tried this format, do you have the same problem in ios?

BigPun86, 2016-10-18:
Well i have an other UUID which is "180A" as a service. But this works on iOS. On Android it crashes when i use this kind of UUID -> "1800"... How does your "common" UUID looks like?

marcosinigaglia, 2016-10-18:
Can you try this change in android?
```java
@ReactMethod
	public void read(String deviceUUID, String serviceUUID, String characteristicUUID, Callback callback) {
		Log.d(LOG_TAG, "Read from: " + deviceUUID);
		Peripheral peripheral = peripherals.get(deviceUUID);
		if (peripheral != null){
			peripheral.read(ParcelUuid.fromString(serviceUUID).getUuid(), UUID.fromString(characteristicUUID), callback);
		} else
			callback.invoke("Peripheral not found", null);
	}
```

joshuar500, 2016-11-28:
@marcosinigaglia @BigPun86 I'm having the same issue. On iOS it works fine. On Android, it says 'Invalid UUID: ED00'. Using the most recent version of ble-manager with the same code posted above.

johneo, 2017-01-09:
I am having a similar issue with version 2.4.3. I get back device ID in this format `20:91:48:2A:69:20` rather than the usual UUID format with the dashes, which iOS code gets back when discovering the devices. Has anyone else run into this issue?

ibenben, 2017-05-25:
I'm having the same issue.
version:` "react-native-ble-manager": "^4.0.0",`
My code:
` //A peripheral was connected.
	    this.BleManagerConnectPeripheralListener = NativeAppEventEmitter.addListener('BleManagerConnectPeripheral', (args) => {
		    console.log('BleManagerConnectPeripheral:', args);
	    });`

` connect(id) {
        return new Promise( (resolve, reject) =>{
            BleManager.connect(id)
                .then((peripheralInfo) => {
                    console.log('Connected peripheralInfo: ', peripheralInfo);
                    resolve(peripheralInfo);
                })
                .catch((error) => {
                    console.log('Connected error:',error);
                    reject(error);
                });
        });
    }
`
log:
`I/ReactNativeJS( 7024): 'BleManagerDiscoverPeripheral:', { rssi: -47,
I/ReactNativeJS( 7024):   name: 'Qianshan01',
I/ReactNativeJS( 7024):   id: '05:04:03:02:01:00',
I/ReactNativeJS( 7024):   advertising:
I/ReactNativeJS( 7024):    { CDVType: 'ArrayBuffer',
I/ReactNativeJS( 7024):      data: 'AgEFAwMSGAsJUWlhbnNoYW4wMQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=' } }
I/ReactNativeJS( 7024): Scan stopped
I/ReactNativeJS( 7024): 'Connected peripheralInfo: ', { characteristics: [],
I/ReactNativeJS( 7024):   services: [],
I/ReactNativeJS( 7024):   rssi: -47,
I/ReactNativeJS( 7024):   advertising:
I/ReactNativeJS( 7024):    { data: 'AgEFAwMSGAsJUWlhbnNoYW4wMQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=',
I/ReactNativeJS( 7024):      CDVType: 'ArrayBuffer' },
I/ReactNativeJS( 7024):   id: '05:04:03:02:01:00',
I/ReactNativeJS( 7024):   name: 'Qianshan01' }
I/ReactNativeJS( 7024): Notification error
I/ReactNativeJS( 7024): 'BleManagerConnectPeripheral:', { peripheral: '05:04:03:02:01:00' }
`
[image]

marcosinigaglia, 2017-05-25:
Hi @ibenben , you need to call retrieveServices method after the connection.

ibenben, 2017-05-26:
@marcosinigaglia Thanks

ibenben, 2017-06-02:
Hi @marcosinigaglia ,the app run pefect on android 5.0 but the other android version is the same issue like 6.0.1 when call retrieveServices method after the connection.

marcosinigaglia, 2017-06-05:
Try to use the `forceLegacy` flag.

batkhai290499, 2020-12-24:
same issue @@. i using version 4.0.0 . i can not het servicesID and characteristicsID. please help. i was use retrieveServices, forceLagacy. nothing change