innoveit/react-native-ble-manager #61: get Could not find peripheral XXXX error when connect a peripheral

haibinyu, 2016-09-29:
react-native-ble-manager version: 2.0.0

like #60 , when I connect a peripheral by id that getting from getConnectedPeripherals, I got error 
```
Could not find peripheral XXXX
```
I check the code and find that it checks if the id is in the discovered peripherals before connecting to it, but in my scenario, the scan operation is skipped if I find the peripheral in the connected peripherals.

As apple recommend at https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/BestPracticesForInteractingWithARemotePeripheralDevice/BestPracticesForInteractingWithARemotePeripheralDevice.html#//apple_ref/doc/uid/TP40013257-CH6-SW9

when reconnecting a peripheral, it is better to get connected peripherals to check if the peripheral is connected before scan.

phoenecke, 2016-10-05:
I also ran into this.  `getConnectedPeripherals' needs to do `[peripherals addObject:peripheral];` or something.

marcosinigaglia, 2016-11-16:
Fixed in 2.3.0

closed 2016-11-16 by marcosinigaglia