innoveit/react-native-ble-manager #62: Latest Ble Manager (2.0.0) doesnt return any bluetooth devices on iOS

Firaenix, 2016-10-05:
I recently updated to version 2.0.0 and when scanning for devices on iOS, I get no devices back.
However, on the same version, on Android, it works correctly.

In the mean time, I've rolled back to version 1.4.0.

I hope this can get fixed!

Thanks very much

marcosinigaglia, 2016-10-05:
Hi, are you sure you have clean the build etc.?

marcosinigaglia, 2016-10-06:
Have you call the init method?

Firaenix, 2016-10-06:
Sorry about the delayed response.
I have not changed any code between version 1.4.0 and version 2.0.0.

I cleaned the build multiple times and also rebuilt the node_modules.

I do not have BleManager.start() in my code, but even so, the android counterpart works as expected, but does not for iOS.

marcosinigaglia, 2016-10-06:
BleManager.start() is added in 2.0.0 and init the module, is mandatory in iOS, in Android is not possible to delay to init so it works without the start method.

closed 2016-10-06 by marcosinigaglia