innoveit/react-native-ble-manager #64: BleManager.enableBluetooth() crashes on Android

BigPun86, 2016-10-06:
Hey there, i am trying to use enableBluetooth() on Android on a real device and it always crashes. 

My stacktrace: 

[image]

Any hints on this how to fix it?

marcosinigaglia, 2016-10-06:
Hi, I can't test on real device now, try replace the enable method with:
```java
	@ReactMethod
	public void enableBluetooth(Callback callback) {
		if (getBluetoothAdapter() == null) {
			Log.d(LOG_TAG, "No bluetooth support");
			callback.invoke("No bluetooth support");
			return;
		}
		if (!getBluetoothAdapter().isEnabled()) {
			enableBluetoothCallback = callback;
			Intent intentEnable = new Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE);
			getCurrentActivity().startActivityForResult(intentEnable, ENABLE_REQUEST);
			callback.invoke();
		} else
			callback.invoke();
	}
```

BigPun86, 2016-10-07:
Nice! Works better now. Atleast it doesn´t crash as soon as i call the function but unfortunately it crashes as soon as i abort the dialog, i deny the dialog or after i have connected to bluetooth.

**"The callback enableBluetooth() exists in module BleManager, but only one callback may be registered to a function in a native module."**

marcosinigaglia, 2016-10-10:
Hi, version [2.1.3](https://github.com/innoveit/react-native-ble-manager/releases/tag/v2.1.3) fix the error.

closed 2016-10-10 by marcosinigaglia

ankushShetty1993, 2021-03-08:
App still crashes on Android when clicked on deny or enable(verison- 7.4.1)