innoveit/react-native-ble-manager #70: BleManagerDidUpdateValueForCharacteristic is never called

BigPun86, 2016-10-17:
Hey there, i need to read the value of an characteristic (in this case a private service !?) 
One problem is, when i read() nothing happens. I tried to trace the native code on ios as well as on android. This is how my read looks like.

First example is when i read "public services":
```
let peripheralId = "some peripheral UUID or MAC address";
let serviceUUID = "180A";
let characteristicUUID = "2A25";
BleManager.read(peripheralId, serviceUUID, characteristicUUID).then((readData) => {
			console.log('value', readData);
			that.hex2a(readData); -> Response is Bluetooth Manufacturer in this case: Mircochip
		}).catch((error) => {
			console.log('err', error);
		});
```

Second example is when i read from "private services":
```
let peripheralId = "some peripheral UUID or MAC address";
let serviceUUID = "00035B03-58E6-07DD-021A-08123A000300";
let characteristicUUID = "00035B03-58E6-07DD-021A-08123A000301";
BleManager.read(peripheralId, serviceUUID, characteristicUUID).then((readData) => {
			console.log('value', readData);
			that.hex2a(readData);
		}).catch((error) => {
			console.log('err', error);
		});
```

Nothing happens when i try the second example...I am not sure what the problem is...maybe the UUID´s are too long :P

marcosinigaglia, 2016-10-19:
Hi, not every characteristic is readable, check the device documentation.

closed 2016-10-19 by marcosinigaglia

jariwalabhavesh, 2017-08-05:
@marcosinigaglia: in my case BleManagerDidUpdateValueForCharacteristic is also not triggering my perticular characteristic of device is readable and notify.

When **BleManagerDidUpdateValueForCharacteristic** actually trigger. do we need to set or configure something to get callback in BleManagerDidUpdateValueForCharacteristic method?