innoveit/react-native-ble-manager #77: Can't discovery to some devices

zhanguangao, 2016-11-22:
I can't discovery to some devices,and the system settings can discovery it.

marcosinigaglia, 2016-11-22:
Hi, can you give more information? Are the devices paired? OS?

zhanguangao, 2016-11-23:
@marcosinigaglia I encounter this problemIn in android. That is I can find the device in the management of the bluetooth in the android Settings, while I can't find it in use this code.

marcosinigaglia, 2016-11-23:
Are ble devices?

zhanguangao, 2016-11-23:
@marcosinigaglia Yes！

marcosinigaglia, 2016-11-23:
Try with getConnectedPeripherals, I have no idea, the scan method is very simple.

zhanguangao, 2016-11-24:
I try ,but have no effect!

marcosinigaglia, 2016-11-24:
Can you paste the code?

zhanguangao, 2016-11-25:
import React, { Component } from 'react';
import {
    AppRegistry,
    Text,
    View,
    StyleSheet,
    TouchableOpacity,
    NativeAppEventEmitter
} from 'react-native';
import BleManager from 'react-native-ble-manager';
//19680   01:C2:3E:F7:62:60
export default class BleExample extends Component {

    constructor(){
        super();
        this.state = {
            ble:null,
            scanning:false,
        }
    }

    componentDidMount() {
        BleManager.start({showAlert: false});

        //The scanning for peripherals is ended.
        NativeAppEventEmitter.addListener('BleManagerStopScan', () => {
            console.log('BleManagerStopScan','Scanning is stopped');
            this.setState({scanning:false, ble: null})
            this.stopScan();
        });

        //The BLE change state.
        NativeAppEventEmitter.addListener('BleManagerDidUpdateState', (args) => {
            console.log('BleManagerDidUpdateStatea',rgs);
        });

        //The scanning find a new peripheral.
        NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', (data) => {
            console.log('BleManagerDiscoverPeripheral',data);
            this.setState({ ble: data });
        });

        //A characteristic notify a new value.
        NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', (args) => {
            console.log('BleManagerDidUpdateValueForCharacteristic',args);
        });

        //A peripheral was connected.
        NativeAppEventEmitter.addListener('BleManagerConnectPeripheral', (args) => {
            console.log('BleManagerConnectPeripheral',args);
        });

        //A peripheral was disconnected.
        NativeAppEventEmitter.addListener('BleManagerDisconnectPeripheral', (args) => {
            console.log('BleManagerDisconnectPeripheral',args);
        });
    }

    connect(){
        BleManager.connect("4B:29:7F:21:B6:FD")
            .then((peripheralInfo) => {
                console.log('Connected');
                console.log(peripheralInfo);
            })
            .catch((error) => {
                console.log(error);
            });
    }

    disconnect(){
[... 113 lines cut ...]
        backgroundColor: '#F5FCFF',
    },
    center:{
        justifyContent: 'center',
        alignItems: 'center',
    },
    button:{
        height:50,
        paddingLeft:20,
        paddingRight:20,
        marginLeft:30,
        marginRight:30,
        borderRadius:5,
        marginTop:20,
        backgroundColor:'#ccc',
    },
    text:{
        fontSize:20,
    },
    list:{
        marginTop:30,
        paddingLeft:20,
        paddingRight:20,
    }
});

marcosinigaglia, 2016-11-28:
Hi, I don't see any error, is the BleManagerDiscoverPeripheral event never call?

zhanguangao, 2016-11-28:
I call it in the following code.
NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', (data) => {
        console.log('BleManagerDiscoverPeripheral',data);
        this.setState({ ble: data });
    });

marcosinigaglia, 2016-11-29:
Don't you see anything on the RN log or iOS console?

joshuar500, 2016-11-29:
I am having the same issue. The correct number of items/devices are displaying correctly in iOS but for some reason Android only gets a very small list of them. It says that there are 50+ devices discovered when there should be only 10. The return on Android is 50+ duplicates of 2-3 devices.

marcosinigaglia, 2016-11-30:
It's not an issue, Android handle the scan in this way, you have to handle manually the duplicates.
http://stackoverflow.com/questions/19502853/android-4-3-ble-filtering-behaviour-of-startlescan/20196114#20196114

joshuar500, 2016-11-30:
That's fine if there are duplicates then. I am saying that the items returned are not the complete list. On iOS, it returns a complete list of items discovered. On Android, the returned list is incomplete, missing over 25 devices that should have been discovered. 

I am getting about the same amount of characters in the hex string (about 1000 chars) on both devices where the string is duplicated, however, Android duplicates a very small portion of the string whereas iOS duplicates a large portion of the string til it equals 1000 chars.

Maybe it's a buffering issue? Anyway to create a function that parses data or a packet every time a response is received?

marcosinigaglia, 2016-11-30:
Do you filter the scan for a specific service? Do you read the data from a characteristics?

joshuar500, 2016-11-30:
Yeah, I can connect to my device fine. The device I'm connected to is broadcasting a list of items (in this case, a list of WiFi access points). I know it's not the device I'm connected to that is the problem. The problem is with Android as I have tried 3 different iPhones and 3 different Android and they all do the same thing. Also using Lightblue for iOS and BLE Scanner for Android to make sure it IS Android.

marcosinigaglia, 2016-12-01:
Try do add a delay betweend the connection and the read, I don't know why but sometimes resolve communication problem. Do you have the same problems using the BLE Scanner?

joshuar500, 2016-12-01:
Yes, same problem on ble scanner

closed 2016-12-01 by marcosinigaglia

ibenben, 2017-04-26:
Yes, same problem on ble scanner.How can I fix it.

ibenben, 2017-04-26:
I am having the same issue. I can get a list of available bluetooth devices using Android bluetooth manager.But I can not discover any avaiable bluetooth device using react-native-ble-manager.

ibenben, 2017-04-26:
@zhanguangao 你这个问题是怎么解决的？能把解决方案告诉我一下吗？

zhanguangao, 2017-04-26:
@ibenben react-native-ble-manager 这个组件只能搜索到低功耗蓝牙设备，像手机的蓝牙即使是打开的也是搜索不到的，而系统内置的蓝牙管理器既可以搜索到普通蓝牙设备和低功耗蓝牙设备

ibenben, 2017-04-26:
@zhanguangao 手机上的不也是ble的吗。还有我的音箱、耳机也搜索不到。你有其它的联系方式吗？

zhanguangao, 2017-04-26:
@ibenben 可以加我微信Guangao_Zhan

WWEISONG, 2020-12-10:
@zhanguangao 请问最后解决了吗

zhanguangao, 2020-12-11:
@WWEISONG react-native-ble-manager只适用于BLE蓝牙通信，不适用于经典蓝牙通信，接入前请确保你的设备是BLE蓝牙。android平台下可以用BlueLoupe软件查看当前蓝牙是否是低功耗的，当有Bluetooth Low Energy 值时，表示当前蓝牙是BLE蓝牙。