innoveit/react-native-ble-manager #79: How to handle Indications (Question)

ekaratzaferis, 2016-11-24:
Hi,

I'm trying to connect to my Glucose meter device and read my latest measurements. I can discover/connect to the device using this library. My device supports the Bluetooth standard about the Glucose devices (I can see the right service uuid's when I connect to the device).

After I have successfully connect my device to my app, 

`BleManager.connect(mac_adress)
        .then((peripheralInfo) => {
               console.log(peripheralInfo);
        })
        .catch((error) => {
                // Failure code
                console.log(error);
        });`

I'm enabling the notifications on the Glucose Measurement characteristic,

`BleManager.startNotification(mac, "00001808-0000-1000-8000-00805F9B34FB", "00002a18-0000-1000-8000-00805F9B34FB")
            .then(() => {
                // Success code
                console.log('GLUCOSE MEASUREMENT NOTIFICATION STARTED');
            })
            .catch((error) => {
                // Failure code
                console.log(error);
            });
`

and then I'm supposed to enable the indications on the RECORD ACCESS CONTROL POINT characteristic, and then follow up by writing some data to the same characteristic that will inform the device on how many records I want to read for example.

Currently I have no idea how to enable indications on this service/characteristic. I've tried using the startNotification method again, but I get this error: 

> Failed to set client characteristic notification for 00002a52-0000-1000-8000-00805f9b34fb

Do you know have I can enable indications on a BLE device? To be honest I'm not 100% sure that the steps are the ones that I have to follow in order to read the measurements, but I can't even try this method.

marcosinigaglia, 2016-11-24:
Hi, the first startNotification works? You have to handle the BleManagerDidUpdateValueForCharacteristic event to read data from the device after you start the notification on some characteristic.

ekaratzaferis, 2016-11-24:
I do manage to start notifications, both on the GLUCOSE MEASUREMENT characteristic and the RECORD ACCESS CONTROL POINT (RACP) characteristic. But no events are triggered.

I've tried writing first to the RACP before starting the notification, and I do receive some feedback from the device, which indicates that there are no measurements, which is weird because I do see them on the device. 

Also, after writing on the device, and restarting the application, I can no longer connect to it. I have to pair the device with the smartphone again in order to discover it.

I'm sending the data this way:

`var myData = [0x01,0x01]`
`var base64 = require('base64-js')`
`var data = base64.fromByteArray(myData)`

I hope I'm not doing anything wrong in this!

Also, is it possible to start notifications on two different characteristics? If I do manage to receive a proper response from the RACP, do I have to stop this notification before starting the CLUCOSE MEASUREMENT? Can I handle both?

marcosinigaglia, 2016-11-24:
You can start notification on multiple characteristics, in the event parameter you can see which characteristic is writing. I don't know how this device works, have you try to read the value directly from the characteristics?

ekaratzaferis, 2016-11-24:
I'm sorry if I'm giving too many technical information.

Is it OK if I "nest" the startNotification in the promise object? I mean something like this:

`BleManager.startNotification(...).then(() => { console(something)'
             `BleManager.startNotification(...).then(() => { console(something)'
             `}).catch((error) => { console.log(error); })`
`}).catch((error) => { console.log(error); })`

I do get this error when I'm trying to do the above:

> Failed to set client characteristic notification for 00002a34-0000-1000-8000-00805f9b34fb

Also, am I writing the data in the proper way? I'd like to write these two bytes: 0x01 0x01 and the more I'm looking my answer above, the more I think I'm doing this wrong.

marcosinigaglia, 2016-11-24:
The data format is correct, look in the connect response, have the two characteristics the indication property? Do you have the error in the first or second startNotification?

ekaratzaferis, 2016-11-24:
I'm trying to subscribe to these characteristics:

a) characteristic:"2a18"
properties:Object
Notify:"Notify"

b) characteristic:"2a34"
properties:Object
Notify:"Notify"

c) characteristic:"2a52"
properties:Object
Indicate:"Indicate"
Write:"Write"

The error appears when I'm trying to subscribe to the second notification. 

Can I start these notifications without nesting them?
I'll try to change the subscription order and see what happens next.

marcosinigaglia, 2016-11-24:
Try to add some delay after the first startNotification.

ekaratzaferis, 2016-11-24:
OK, that actually worked great! Do you have any idea if this has to do with my BLE device, my smartphone, the React Native Dev Server or the library?

Anyways, I'm able to subscribe to all 3 characteristics and the GLUCOSE METER one, triggered some results. I'm going to decode them and see what's going on.

I guess it's up to my device from this point now.

Thank you so much for your help!

closed 2016-11-24 by ekaratzaferis

marcosinigaglia, 2016-11-24:
I don't know what is the real problem, on my device i had to add some delay too after the notification before to start to send data. I had the same result in iOS and Android so I think is BLE chip that confirm the notification too early or it need some time before others interactions.

stephenjen, 2017-05-17:
@ekaratzaferis I'm trying to do something similar to what you did here, mainly I need to send the device an instruction to start sending data over.  Can you share your code on how you did this?  Not sure how startNotification, write, and BleManagerDidUpdateValueForCharacteristic piece together to make this work.  

Thanks!

ekaratzaferis, 2017-05-20:
Hi and sorry for the delay!

To be honest, after connecting to the device, I gave up on this project, so I haven't implemented any functions for reading/writing data to the device.

Also, the connect method is truly ugly, you should do something about it!

By the way, I was learning React Native + Redux + ECMAScript 6 at the time, so the code is messed up (a lot) :/

```
import BleManager from 'react-native-ble-manager'
import { NativeAppEventEmitter } from 'react-native'

const ble_api = {

    // Variables //
    scanning: {
        active: false,
        interval: null,
    },
    device: {
        found: false,
        info: {
            name: '',
            id: ''    
        }
    },

    // Bind listener for the discovery event //
    initiate: function (discovery_handler) {
        BleManager.start({showAlert: false})
        this.peripheral_found = this.peripheral_found.bind(this, discovery_handler)
        NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.peripheral_found)
    },

    // Check if bluetooth is activated on the device and run the handler //
    is_on: function (handler) {
        BleManager.enableBluetooth().then(() => {
            handler('on')
        }).catch((error) => {
            handler('off')
        })
    },

    // Toggle device scanning //
    toggle_scan: function (bool) {
        if (bool) {
            this.scanning.active = true
            this.device.found = false
            this.scanning.interval = setInterval(() => 
                BleManager.scan([], 30, true).then((results) => this.device.found = true), 3000)
        } else{
            BleManager.stopScan().then(() => this.scanning.active = false)
            clearInterval(this.scanning.interval)
        }
    },

    // Execute assigned handler //
    peripheral_found: function (handler,data) {
        // Stop device scanning and connect to the peripheral //
        this.toggle_scan(false)
        this.connect(data.id)
        // Connect to the device //
        this.device.info = data
        // Run handler //
        handler(data)
    },

    // Connect //
    connect: function(mac){
        BleManager.connect(mac)
        .then((peripheralInfo) => {
            // Success code //
            console.log(peripheralInfo);
            BleManager.startNotification(mac, "00001808-0000-1000-8000-00805F9B34FB", "00002a18-0000-1000-8000-00805F9B34FB",)
            .then(() => {
                console.log('CLUCOSE MEASUREMENT NOTIFICATIONS: ON');
                NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', function(data){ 
                    console.log('RESULT FROM CLUCOSE MEASUREMENT NOTIFICATION')
                    console.log(data) 
                })
                setTimeout(function() {
                    BleManager.startNotification(mac, "00001808-0000-1000-8000-00805F9B34FB", "00002a34-0000-1000-8000-00805F9B34FB")
                    .then(() => {
                        console.log('CLUCOSE MEASUREMENT CONTEXT NOTIFICATIONS: ON');
                        NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', function(data){ 
                            console.log('RESULT FROM CLUCOSE MEASUREMENT CONTEXT NOTIFICATION')
                            console.log(data)
                        })
                        setTimeout(function() {
                            BleManager.startNotification(mac, "00001808-0000-1000-8000-00805F9B34FB", "00002a52-0000-1000-8000-00805F9B34FB")
                            .then(() => {
                                console.log('SUBSCRIBING TO RAPC')
                                NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', function(data){ 
                                    console.log('RESULT FROM RAPC NOTIFICATION')
                                    console.log(data)
                                });
                                setTimeout(function() {
                                    var myData = [0x01,0x01]
                                    var base64 = require('base64-js')
                                    var data = base64.fromByteArray(myData)
                                    BleManager.write(mac, "00001808-0000-1000-8000-00805F9B34FB", "00002a52-0000-1000-8000-00805F9B34FB",data)
                                    .then(() => {
                                        console.log('Write' + data)
                                    }).catch((error) => { console.log(error) })
                                }, 3000); 
                            }).catch((error) => { console.log(error) });
                        }, 3000); 
                    }).catch((error) => { console.log(error) })
                }, 3000);
            }).catch((error) => { console.log(error) })
        }).catch((error) => { console.log(error) })
    },

    // Read measurement from device //
    read: function(){
    
    },

    // Read all measurements from device //
    read_all: function(){
        
    },
    // Send data to the device //
    write: function(){
        
    }
};
module.exports = ble_api
```

If I recall correctly my train of thought was something like this:
Use **is_on** to check if bluetooth is turned on the device and then use **initiate**
I had a button assigned to toggle device scanning with **toggle_scan**
**peripheral_found** was executed upon discovery
**connect** to the device by adding some ugly intervals between the "commands"

I did get some data back from the device by I didn't bother to continue at the time.

Hope this gives you an idea on how to write your API, and maybe share your work with others too :)

stephenjen, 2017-05-26:
@ekaratzaferis Thanks for posting your code.  I had already figured out that I needed to put NativeAppEventEmitter.addListener in the callback of BleManager.startNotification, which was the reason why mine wasn't getting triggered.

Utkarshlakhera, 2022-01-22:
@stephenjen Your above issue fixed? if yes, so can you tell me that how we can resolve it. I am getting issue in the android device. Failed to set client characteristic notification for "charID(xyz)".