innoveit/react-native-ble-manager #83: Getting error code- 128 while writing to characterstic

shubhapp, 2016-12-07:


marcosinigaglia, 2016-12-08:
I think is a device or Android stack problem. Can you give me some other info?

shubhapp, 2016-12-10:
I don't think , the problem is with the device or android stack because if am using some other app like [BleScanner](https://play.google.com/store/apps/details?id=com.macdom.ble.blescanner&hl=en) to perform write operation it is working fine. I am not sure but I think the problem is with the value which I am trying to write to the Ble device. Do I need to modify value some way apart from converting it into base64

marcosinigaglia, 2016-12-12:
The -128 is GATT_NO_RESOURCES, i don't think is value problem. Have you tried write with response and without response?

shubhapp, 2016-12-13:
I have tried both 'write with response and without response' . In case of without response i am not getting any error. it is returning success. But in case of with response ,I always get 128 error.

marcosinigaglia, 2016-12-13:
Are you sure the characteristic support write without response?

shubhapp, 2016-12-26:
Hi This is characteristic property i am getting --
characteristic:
uuid:"e16c6e20-b041-11e4-a4c3-0002a5d5c51b"
isNotifiable:true
isReadable:false
isWritable:true
isWriteWithoutResponse:false

It is clearly indicating that characteristic is not WriteWithoutResponse.

shubhapp, 2016-12-28:
@marcosinigaglia can you provide some help on this ?

marcosinigaglia, 2017-01-03:
Did the device's documentation have any information on this error? Can you read the device log for others informations?

shubhapp, 2017-01-04:
as per my understanding device log you are referring to ble device logs, In that case,  I do not have access to device logs as it is third party device.

marcosinigaglia, 2017-01-04:
I mean the android's log

shubhapp, 2017-01-05:
These are the logs i am getting-

`01-05 16:07:20.779 14606-23130/com.careconnect D/logs: scan
01-05 16:07:20.780 14606-23130/com.careconnect D/logs: Filter service: daebb240-b041-11e4-9e45-0002a5d5c51b
01-05 16:07:20.782 14606-23130/com.careconnect D/BluetoothAdapter: STATE_ON
01-05 16:07:20.785 14606-14864/com.careconnect D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
01-05 16:07:27.507 14606-14606/com.careconnect I/logs: DiscoverPeripheral: My IQOS 2.4+
01-05 16:07:27.564 14606-23130/com.careconnect D/logs: Stop scan
01-05 16:07:27.566 14606-23130/com.careconnect D/BluetoothAdapter: STATE_ON
01-05 16:07:27.566 14606-23130/com.careconnect D/BluetoothLeScanner: could not find callback wrapper
01-05 16:07:27.638 14606-23130/com.careconnect D/logs: Connect to: CE:AD:F1:A1:5D:00
01-05 16:07:27.643 14606-23130/com.careconnect D/BluetoothGatt: connect() - device: CE:AD:F1:A1:5D:00, auto: false
01-05 16:07:27.643 14606-23130/com.careconnect D/BluetoothGatt: registerApp()
01-05 16:07:27.643 14606-23130/com.careconnect D/BluetoothGatt: registerApp() - UUID=5dc474ac-2a21-49d9-a7ae-aeef687f6b6b
01-05 16:07:27.645 14606-14618/com.careconnect D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
01-05 16:07:27.678 14606-23130/com.careconnect D/logs: Stop scan
01-05 16:07:27.679 14606-23130/com.careconnect D/BluetoothAdapter: STATE_ON
01-05 16:07:27.679 14606-23130/com.careconnect D/BluetoothLeScanner: could not find callback wrapper
01-05 16:07:28.789 14606-14618/com.careconnect D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=CE:AD:F1:A1:5D:00
01-05 16:07:28.789 14606-14618/com.careconnect D/logs: onConnectionStateChange from 0 to 2 on peripheral:CE:AD:F1:A1:5D:00
01-05 16:07:28.789 14606-14618/com.careconnect D/BluetoothGatt: discoverServices() - device: CE:AD:F1:A1:5D:00
01-05 16:07:28.793 14606-14618/com.careconnect D/logs: Peripheral event (eventName):CE:AD:F1:A1:5D:00
01-05 16:07:30.838 14606-14618/com.careconnect D/BluetoothGatt: onSearchComplete() = Device=CE:AD:F1:A1:5D:00 Status=0
01-05 16:07:30.962 14606-23130/com.careconnect D/logs: startNotification
01-05 16:07:30.962 14606-23130/com.careconnect D/logs: registerNotify
01-05 16:07:30.962 14606-23130/com.careconnect D/logs: setNotify
01-05 16:07:30.963 14606-23130/com.careconnect D/BluetoothGatt: setCharacteristicNotification() - uuid: e16c6e20-b041-11e4-a4c3-0002a5d5c51b enable: true
01-05 16:07:30.964 14606-23130/com.careconnect D/logs: Characteristic e16c6e20-b041-11e4-a4c3-0002a5d5c51b set NOTIFY
01-05 16:07:30.964 14606-23130/com.careconnect D/logs: setNotify complete
01-05 16:07:33.074 14606-23130/com.careconnect D/logs: checkState
01-05 16:07:33.075 14606-23130/com.careconnect D/logs: state:on
01-05 16:07:33.078 14606-23130/com.careconnect D/logs: Write to: CE:AD:F1:A1:5D:00
01-05 16:07:33.078 14606-23130/com.careconnect D/logs: Message(7): D340B4D35DB417
01-05 16:07:33.083 14606-23130/com.careconnect D/logs: Write completed
01-05 16:07:33.175 14606-14618/com.careconnect E/logs: Error onCharacteristicWrite:128
01-05 16:07:33.199 14606-23129/com.careconnect I/ReactNativeJS: { size: 1,
                                                                  _origin: 0,
                                                                  _capacity: 1,
                                                                  _level: 5,
                                                                  _root: null,
                                                                  _tail: { array: [ 'Error writing status: 128' ], ownerID: {} },
                                                                  __ownerID: undefined,
                                                                  __hash: undefined,
                                                                  __altered: true }
01-05 16:07:33.260 14606-23130/com.careconnect D/logs: Disconnect from: CE:AD:F1:A1:5D:00
01-05 16:07:33.260 14606-23130/com.careconnect D/BluetoothGatt: cancelOpen() - device: CE:AD:F1:A1:5D:00
01-05 16:07:33.261 14606-23130/com.careconnect D/BluetoothGatt: close()
01-05 16:07:33.261 14606-23130/com.careconnect D/BluetoothGatt: unregisterApp() - mClientIf=6
01-05 16:07:33.263 14606-23130/com.careconnect D/logs: Disconnect
01-05 16:07:33.263 14606-23130/com.careconnect D/logs: Peripheral event (eventName):CE:AD:F1:A1:5D:00
01-05 16:07:50.802 14606-14606/com.careconnect D/BluetoothAdapter: STATE_ON
`

marcosinigaglia, 2017-01-05:
Try to add some delay between the connection and the start notification.

shubhapp, 2017-01-06:
These are the new logs after adding delay. I have added 2000miliseconds delay between connect and startNotification , startNotification and write characteristic.
 
01-06 11:03:03.099 24401-24440/com.test D/logs: scan
01-06 11:03:03.102 24401-24440/com.test D/logs: Filter service: daebb240-b041-11e4-9e45-0002a5d5c51b
01-06 11:03:03.103 24401-24440/com.test D/BluetoothAdapter: STATE_ON
01-06 11:03:03.108 24401-24414/com.test D/BluetoothLeScanner: onClientRegistered() - status=0 clientIf=5
01-06 11:03:03.625 24401-24401/com.test I/logs: DiscoverPeripheral: My Device
01-06 11:03:03.674 24401-24440/com.test D/logs: Stop scan
01-06 11:03:03.677 24401-24440/com.test D/BluetoothAdapter: STATE_ON
01-06 11:03:03.677 24401-24440/com.test D/BluetoothLeScanner: could not find callback wrapper
01-06 11:03:03.781 24401-24440/com.test D/logs: Connect to: CE:AD:F1:A1:5D:00
01-06 11:03:03.782 24401-24440/com.test D/BluetoothGatt: connect() - device: CE:AD:F1:A1:5D:00, auto: false
01-06 11:03:03.782 24401-24440/com.test D/BluetoothGatt: registerApp()
01-06 11:03:03.782 24401-24440/com.test D/BluetoothGatt: registerApp() - UUID=fb5fa31e-6d72-4e90-a25f-e823b3faff70
01-06 11:03:03.784 24401-24713/com.test D/BluetoothGatt: onClientRegistered() - status=0 clientIf=6
01-06 11:03:03.826 24401-24440/com.test D/logs: Stop scan
01-06 11:03:03.827 24401-24440/com.test D/BluetoothAdapter: STATE_ON
01-06 11:03:03.827 24401-24440/com.test D/BluetoothLeScanner: could not find callback wrapper
01-06 11:03:04.076 24401-24412/com.test D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=6 device=CE:AD:F1:A1:5D:00
01-06 11:03:04.077 24401-24412/com.test D/logs: onConnectionStateChange from 0 to 2 on peripheral:CE:AD:F1:A1:5D:00
01-06 11:03:04.077 24401-24412/com.test D/BluetoothGatt: discoverServices() - device: CE:AD:F1:A1:5D:00
01-06 11:03:04.080 24401-24412/com.test D/logs: Peripheral event (eventName):CE:AD:F1:A1:5D:00
01-06 11:03:04.089 24401-24414/com.test D/BluetoothGatt: onSearchComplete() = Device=CE:AD:F1:A1:5D:00 Status=0
01-06 11:03:06.214 24401-24440/com.test D/logs: startNotification
01-06 11:03:06.214 24401-24440/com.test D/logs: registerNotify
01-06 11:03:06.214 24401-24440/com.test D/logs: setNotify
01-06 11:03:06.214 24401-24440/com.test D/BluetoothGatt: setCharacteristicNotification() - uuid: f8a54120-b041-11e4-9be7-0002a5d5c51b enable: true
01-06 11:03:06.215 24401-24440/com.test D/logs: Characteristic f8a54120-b041-11e4-9be7-0002a5d5c51b set NOTIFY
01-06 11:03:06.215 24401-24440/com.test D/logs: setNotify complete
01-06 11:03:08.278 24401-24440/com.test D/logs: checkState
01-06 11:03:08.279 24401-24440/com.test D/logs: state:on
01-06 11:03:08.285 24401-24440/com.test D/logs: Write to: CE:AD:F1:A1:5D:00
01-06 11:03:08.285 24401-24440/com.test D/logs: Message(5): 00C00120F5
01-06 11:03:08.286 24401-24440/com.test D/logs: Write completed
01-06 11:03:08.363 24401-24713/com.test E/logs: Error onCharacteristicWrite:128
01-06 11:03:08.448 24401-24440/com.test D/logs: Disconnect from: CE:AD:F1:A1:5D:00
01-06 11:03:08.448 24401-24440/com.test D/BluetoothGatt: cancelOpen() - device: CE:AD:F1:A1:5D:00
01-06 11:03:08.450 24401-24440/com.test D/BluetoothGatt: close()
01-06 11:03:08.450 24401-24440/com.test D/BluetoothGatt: unregisterApp() - mClientIf=6
01-06 11:03:08.452 24401-24440/com.test D/logs: Disconnect
01-06 11:03:08.452 24401-24440/com.test D/logs: Peripheral event (eventName):CE:AD:F1:A1:5D:00

shubhapp, 2017-01-06:
`
connectPeripheral = (selectedPeripheralId) => {
    this.props.changePeripheralState(selectedPeripheralId, PERIPHERAL_CONNECTION_STATE.CONNECTING);
    BleManager.connect(selectedPeripheralId)
    .then((peripheralInfo) => {
      this.onPeripheralConnected(selectedPeripheralId, peripheralInfo);
    })
    .catch((error) => {
      this.props.pushError(error);
    });
  };

    onPeripheralConnected = (selectedPeripheralId, peripheralInfo) => {
    const servicesMap = mapServicesAndCharacteristics(peripheralInfo.services, peripheralInfo.characteristics);
    this.props.updateServices(selectedPeripheralId, servicesMap);
    setTimeout(() => {
            this.startCharacteristicNotification(newProps.selectedPeripheralId, primaryService.UUID, primaryService.BATTERY_INFO.uuid, true);
          }, 2000);
  };

startCharacteristicNotification=(peripheralId, serviceId, characteristicUUID, transactionId) => {
    BleManager.startNotification(peripheralId, serviceId, characteristicUUID)
     .then(() => {
              this.props.updateCharacteristic(peripheralId, serviceId, characteristicUUID, { isNotifying: true });
              this.props.completeTransaction(transactionId);

              setTimeout(() => {
                const base64String = new Buffer('00C00120F5', 'hex').toString('base64');
                this.props.writeCharacteristic(this.props.selectedPeripheralId, primaryService.UUID, primaryService.CHARACTERISTICS.uuid, base64String);
              }, 2000);
            })
            .catch((error) => {
              this.props.pushError(error);
              this.props.completeTransaction(transactionId);
            });
  };

    writeCharacteristicValue = (peripheralId, serviceId, characteristicUUID, base64Value, transactionId) => {
    BleManager.checkState();
    this.props.executeTransaction(transactionId);
    BleManager.write(peripheralId, serviceId, characteristicUUID, base64Value)
    .then(() => {
      this.props.completeTransaction(transactionId);
      this.props.updateCharacteristic(peripheralId, serviceId, characteristicUUID, { value: base64Value });
    })
    .catch((error) => {
      this.props.pushError(error);
      this.props.completeTransaction(transactionId);
    });
  };`

marcosinigaglia, 2017-01-09:
Hi, what are the differences? The characterics are different in the two logs, is it correct?

shubhapp, 2017-01-10:
I just changed service uuid (random)while posting it on Github.

marcosinigaglia, 2017-01-10:
What has changed in the two logs?

shubhapp, 2017-01-11:
I think, I didn't get you. What changes we are talking about. I haven't changed anything in code apart from adding delay. Just copied the logs which I got from android logcat.

marcosinigaglia, 2017-01-16:
Ok sorry, I don't see any problem in your code, can you contact the BLE device's vendor for some hint?

shubhapp, 2017-01-17:
When i am trying to convert hex to byte array using below code . It is working 

`public static byte[] hexStringToByteArray(String s) {
    int len = s.length();
    byte[] data = new byte[len / 2];
    for (int i = 0; i < len; i += 2) {
        data[i / 2] = (byte) ((Character.digit(s.charAt(i), 16) << 4)
                             + Character.digit(s.charAt(i+1), 16));
    }
    return data;
}`

But i am getting another problem after making these changes. When I try to write first time it works but when i try to write again(after 2000 milliseconds) then I get error saying bluetoothGatt is null.

marcosinigaglia, 2017-01-17:
Maybe is the device that close the connection after the first write?

shubhapp, 2017-01-20:
@marcosinigaglia Thanks for help and support. everything is working fine now. 👍

closed 2017-01-20 by marcosinigaglia

gpaz, 2018-07-30:
@marcosinigaglia  Do you have a solution to your problem?  Do you know why it happened?

0xcoax, 2019-09-15:
@shubham0390 i've seen that you are interested on iqos device, how i can contact you?

limneos, 2020-10-30:
@shubham0390 can you please post the exact data bytes that you sent when it worked for you?  I'm trying to do the same on an IQOS 3 Duo