innoveit/react-native-ble-manager #85: Unhandled Promise Rejection

omargallob, 2016-12-09:
We have been trying to use this library on ExponentJS. 

We get warnings with

" Possible Unhandled Promise Rejection (id: 0): undefined is not an object (evaluating 'bleManager.start')"
" Possible Unhandled Promise Rejection (id: 1): undefined is not an object (evaluating 'bleManager.scan')"

We later tried to test the example code provided in the repos readme file on a fresh react native app. Followed the steps regarding Xcode and we get the following output in xcode

2016-12-09 14:05:40.594 [info][tid:main][RCTBatchedBridge.m:72] Initializing <RCTBatchedBridge: 0x1756be60> (parent: <RCTBridge: 0x1766e3e0>, executor: RCTJSCExecutor) 2016-12-09 14:05:40.933 BleExample[3347:1247845] BleManager created 2016-12-09 14:05:41.498 [warn][tid:com.facebook.react.JavaScript][RCTBatchedBridge.m:505] Failed to execute source code: Unhandled JS Exception: SyntaxError 2016-12-09 14:05:41.501 [fatal][tid:main] Unhandled JS Exception: SyntaxError

and red screen of death with 

" Unhandled JS Exception: SyntaxError"

XiaofanFTD, 2016-12-12:
haha, I solved it. 
Maybe you should confirm you had do follow step:
Check the "Build Phases"of your project and add "libBleManager.a" in the "Link Binary With Libraries" section.

marcosinigaglia, 2016-12-12:
Yes, I think is an installation problem.

omargallob, 2016-12-12:
I added the libBleManager.a in xcode, will double check and try again.

will give it another go this evening. 

Kind Regards

marcosinigaglia, 2016-12-12:
Check the Build Phases and try to clean the project.

closed 2016-12-16 by marcosinigaglia

chase1745, 2018-01-28:
Where is the "libBleManager.a"? I can't seem to find it anywhere.