innoveit/react-native-ble-manager #86: Android: connect to peripheral success, but callback object undefined

bardu, 2016-12-14:
I'm using v2.4.2 and my demo code looks like this:

```
handleDiscoverPeripheral = (data) => {
        console.log('Got ble data', data);
        this.setState({ ble: data })
        if ( data.name === gateway.name ) {
            console.log('found gateway!!!!!!!!!!!!!!')
            this.handleConnect(data.id);
            this.setState({scanning:false})
            clearInterval(this.scanning);
        }
    }

handleConnect = (id) => {
        BleManager.connect(id)
            .then((peripheralInfo) => {
                console.log('connected to gateway');
                console.log(peripheralInfo);
                BleManager.getConnectedPeripherals([])
                    .then((peripheralArray) => {
                        console.log('Connected peripherals = ', peripheralArray.length)
                    });
            })
            .catch((error) => {
                console.log('Error connecting to gateway = ', error);
            });
    }
```
I'm able to connect to the peripheral, but the peripheralInfo object is undefined. Checking the connected peripherals seems to be right, peripheralArray[0] is the peripheral Android device is connected to.

Any clues what could be wrong?

bardu, 2016-12-14:
While [nRF Connect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) shows the right 128 bit UUID for services and characteristics, I'm wondering whether this library supports 128 bit vendor UUID?

marcosinigaglia, 2016-12-14:
Maybe is this bug? https://code.google.com/p/android/issues/detail?id=59490&q=BLE&colspec=ID%20Type%20Status%20Owner%20Summary%20Stars
Can you paste your UUID?

bardu, 2016-12-14:
e8ae0002-0d06-48c7-89f8-70450cda7cc8

bardu, 2016-12-14:
@marcosinigaglia  Some more information: I'm testing on a Samsung Galaxy S6 Edge with Android v6.0.1 and since nRF Connect shows the correct service and characteristics I don't think my issue is related to the bug you have mentioned. Here is a screenshot from nRF:

[image]

There are actually 18 characteristics for the service in total.
If you need more information please let me know.

bardu, 2016-12-15:
Another note: Testing the demo code on iPhone 6 with iOS 10.1.1 works. Unfortunately, this is not our target platform.

marcosinigaglia, 2016-12-15:
Is the BleManagerDiscoverPeripheral event returning the peripheral correctly?

marcosinigaglia, 2016-12-15:
I did some tests with no error, you have to debug the module, check the onServicesDiscovered method in Peripheral.java

bardu, 2016-12-15:
@marcosinigaglia  this is what BleManagerDiscoverPeripheral event is returning:

>12-15 09:46:21.886 30496 31382 I ReactNativeJS: found gateway!!!!!!!!!!!!!!
12-15 09:46:21.886 30496 31382 I ReactNativeJS: 'Got ble data', { rssi: -55,
12-15 09:46:21.886 30496 31382 I ReactNativeJS:   name: 'GatewayV1.2',
12-15 09:46:21.886 30496 31382 I ReactNativeJS:   id: 'C6:52:9D:D0:5C:68',
12-15 09:46:21.886 30496 31382 I ReactNativeJS:   advertising: 
12-15 09:46:21.886 30496 31382 I ReactNativeJS:    { CDVType: 'ArrayBuffer',
12-15 09:46:21.886 30496 31382 I ReactNativeJS:      data: 'DAlHYXRld2F5VjEuMgIBBQUDDxgKGBEHyHzaDEVw+InHSAYNAgCu6AAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=' } }

Does that help?

bardu, 2016-12-16:
@marcosinigaglia So, I fixed my issue by changing AndroidManifest.java to

>android:minSdkVersion="21"
        android:targetSdkVersion="23"

When creating a react-native project the value were set to 16 and 19 respective automatically.

closed 2016-12-16 by bardu