innoveit/react-native-ble-manager #87: Can I send hexadecimal data to BLE instead of base64 format?

zhanguangao, 2016-12-15:


zhanguangao, 2016-12-16:
Can I modify the source code to send hexadecimal ?

marcosinigaglia, 2016-12-16:
I don't understand the problem, you can convert you hexadecimal value in base64.

zhanguangao, 2016-12-16:
Because of the BLE device only analysis the hexadecimal  value, so I only to send hexadecimal  value.

marcosinigaglia, 2016-12-16:
The base64 is only the way to send bytes from javascript to the native part of the module, the module reconvert the base64 string in bytes and write to the device.

zhanguangao, 2016-12-17:
I change the source code to send String value, and transform string to byte array as parameter to  peripheral.write(),and transform byte to hexadecimal byte array as parameter to  characteristic.setValue(), finally the BLE device successfully read the value.