innoveit/react-native-ble-manager #89: [Android] connect(peripheralId) does not connect when Bluetooth has been restarted

vikeri, 2016-12-20:
First I'd like to thank you for a great library!

If I have connected to a device and stored its peripheralId and then restart the BLE device I can reconnect by just calling `connect(peripheralId)`. This works for both Android and iOS. But if I turn off the Bluetooth on the Android and then start it again it won't connect using the `connect(peripheralId)` method directly to the device but has to scan, find the device and then connect.

This might be an issue in how Android works rather than an issue with the library. But then it should probably be documented.

marcosinigaglia, 2016-12-20:
Hi @vikeri , I think that could happen in iOS too, if you have already the device id/mac address you can try to connect directly but if fail you have scan for the peripheral. I will add it as you suggest in the documentation.

vikeri, 2016-12-20:
Alright, for me it worked on iOS. But then I'll add that special case for Android.

closed 2016-12-20 by marcosinigaglia