innoveit/react-native-ble-manager #91: failed to find target with hash string 'android-24'

viney, 2016-12-28:
exec: react-native run-android

viney@go:~/android/ws/ble$ react-native run-android
Starting JS server...
Running /home/viney/android/sdk/platform-tools/adb -s 3069da3f reverse tcp:8081 tcp:8081
Building and installing the app on the device (cd android && ./gradlew installDebug)...

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':app'.
> A problem occurred configuring project ':react-native-ble-manager'.
   > failed to find target with hash string 'android-24' in: /home/viney/android/sdk

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 3.027 secs
Could not install the app on the device, read the error above for details.
Make sure you have an Android emulator running or a device connected and have
set up your Android development environment:
https://facebook.github.io/react-native/docs/android-setup.html

marcosinigaglia, 2017-01-03:
I think you have to install the android 24 framework.

closed 2017-01-10 by marcosinigaglia

anwarhamr, 2017-08-24:
Does this mean that this package will only work w/ Nougat or greater?