innoveit/react-native-ble-manager #92: Often  bluetoothgatt is null

a709691139, 2017-01-03:
First open the app, successfully connected bluetooth,
then close the app, reopen app,    bluetoothgatt is null,
To click several times in order to be successful

marcosinigaglia, 2017-01-03:
Hi, is the device still connected?

a709691139, 2017-01-04:
close the app,  Already disconnected

a709691139, 2017-01-04:
Hope there is a function :  restart bluetooth

closed 2017-01-09 by a709691139

marcosinigaglia, 2017-01-09:
Have you solve the problem?

a709691139, 2017-01-10:
I used this method ： When I connect to Bluetooth,  i save it。 
After re opening,  use the "disconnect and stopNotification"

shubhapp, 2017-01-17:
When i am trying to convert hex to byte array using below code . It is working 
`byte[] getBytes(String s){
        int len = s.length();
        byte[] data = new byte[len / 2];
        for (int i = 0; i < len; i += 2) {
            data[i / 2] = (byte) ((Character.digit(s.charAt(i), 16) << 4)
                    + Character.digit(s.charAt(i+1), 16));
        }
        return data;
    }`