innoveit/react-native-ble-manager #95: How to get the Device name when BleManager discover a new one?

zphhhhh, 2017-01-06:
I'm developing an Android app and using React Native 0.39 and react-native-ble-manager 2.4.3, testing on an Android 6.0 device. And I meet a problem :

The example gived below the event 'BleManagerDiscoverPeripheral' in Readme is :
`
NativeAppEventEmitter.addListener(
    'BleManagerDiscoverPeripheral',
    (args) => {
        // The id: args.id
        // The name: args.name
    }
);
`
but the args.name is undefined, the args is like :

`
// console.log(JSON.stringify(args))
{
  rssi: -64,
  id: '6D-7E-xx-xx-xx-xx-xx-xx',
  advertising: {
    CDVType: 'ArrayBuffer',
    data: 'AgE/adsAAAAAAAAAAAZXXXXXXXXXXX='
  }
}
`
but how can i get the device name? 
I have not developed Android before. Thank you very much!

marcosinigaglia, 2017-01-09:
Hi, I think the ble device doesn't have a name.  I'm not an expert on BLE stack but i think you have to connect to the device even you are paired.

zphhhhh, 2017-01-11:
Thank you, I have solved the problem!
I can get the device's name after connecting to the device. First to pair, then to connect.

But I think the methods this BleManager provides is less than those provided by Native Android API, such as BluetoothAdapter.getBondedDevices().

Thank you again!

closed 2017-01-11 by zphhhhh