innoveit/react-native-ble-manager #97: Could not find service with UUID on peripheral with UUID

joshuar500, 2017-01-12:
I am trying to make a call on a read and it works the first time, but if I unexpectedly unmount the component that is in the middle of receiving the call, and try to make the read call again, I get this error:

`Could not find service with UUID 967C8518-A801-4D25-B215-34C30176346A on peripheral with UUID 9ED2B473-1899-4C82-AF06-1837751F291E`

When I am connecting the first time and it is a success, this is what it says:
```
Looking for 2751DB6D-C940-4F2A-8AFA-07B56C7C414E with properties 2
Found 2751DB6D-C940-4F2A-8AFA-07B56C7C414E
```

The catch happens in here from the first function:
```
BleManager.read(this.state.peripheralUUID,
      this.state.serviceUUID,
      this.state.networkListCharacteristicUUID).then((response) => {
        this.readResponse(this.state.networkListCharacteristicUUID)
      })
      .catch((error) => {
        console.log(error)
      })
```

And `readResponse` is another function doing another read. But this is not the function that is throwing the error however I wonder if it has anything to do with it...

```
readResponse (UUIDToRead) {
    this.setState({outputLog: 'Reading response'})
    BleManager.read(this.state.peripheralUUID,
      this.state.serviceUUID,
      UUIDToRead).then((response) => {
        if (this.state.currentCommand === this.state.commands[0]) {
          const strVal = hex2a(response)
          var networks = strVal.split(', ')
          // remove duplicates
          let uniqueNetworks = [...new Set(networks)]
          this.setState({wirelessNetworks: uniqueNetworks})
          this.setState({
            dataSource: this.state.dataSource.cloneWithRows(uniqueNetworks)
          })
          if (Platform.OS === 'android') {
            // may need this regardless, if we can't solve issue #11
            this.setState({
              outputLog: '',
              isAndroid: true,
              androidHelper: 'Don\'t see your network but know the name?' +
              '\nType it in the first empty box below.'
            })
          } else {
            this.setState({outputLog: 'Please select your network.'})
          }
        } else if (this.state.currentCommand === this.state.commands[1]) {
          const ipAddress = hex2a(response)
          this.setState({outputLog: this.state.outputLog + '\n ip addy: ' + ipAddress + ' ' + response})
          if (ipAddress.length === 0) {
            this.setState({currentCommand: this.state.commands[0]})
          } else {
            this.setState({currentCommand: this.state.commands[2]})
          }
          if (this.state.currentCommand === this.state.commands[0]) {
            Toast.showLongBottom('Couldn\'t connect to ' + this.state.wirelessSSID + '. Please try again')
          } else {
            Toast.showLongBottom('Connected to ' + ipAddress)
            this.setState({apiIPAddress: ipAddress})
            AsyncStorage.setItem('apiIPAddress', ipAddress).done()
            this.setState({
              dataSource: this.state.dataSource.cloneWithRows([])
            })
            AppConfig.trntblIPAddress = ipAddress
            BleManager.disconnect(this.state.peripheralUUID)
            NavigationActions.pop({refresh: {key: 'presentationScreen'}})
          }
        }
      })
      .catch((error) => {
        BleManager.disconnect(this.state.peripheralUUID)
        this.setState({outputLog: this.state.outputLog + '\nUnable to read: ' + error})
        this.setState({attemptingToConnect: false})
      })
  }
```

Just to clarify, this DOES work on the first call, but each additional call if I interrupt/unmount the component, it does not work and I get the aforementioned error. This is what I'm doing for the unmount:

```
componentWillUnmount () {
    this.toggleScanning(false)
    BleManager.stopScan()
    BleManager.disconnect(this.state.peripheralUUID)
    this.nae.remove() // this is BleManagerDiscoverPeripheral
  }
```

Am I missing anything? When I restart the app, it works again the first time, I'm just having issues with it when unmounting the component. Thanks for any help.

marcosinigaglia, 2017-01-17:
Try this solution https://github.com/innoveit/react-native-ble-manager/issues/92

joshuar500, 2017-01-20:
It didn't seem to help. I tried saving the peripheral, service, and characteristic in storage and disconnecting from peripheral and  stopNotification in the app's root screen as well as the componentWillUnmount but neither seemed to work.

tamirla, 2017-05-05:
I'm also getting this error when I'm connecting to one peripheral, then disconnect from it and trying to connect to another one. I see this problem only in iOS. Not sure the root cause is the same as described here but the error is the same. any idea what can cause it in my case ?

marcosinigaglia, 2017-05-05:
Hi @tamirla , explain what do you do, it can be anything.

tamirla, 2017-05-05:
@marcosinigaglia I read again the issue described above, and notice my issue is actually similar. In my app I have 2 pages, in first one I present discovered peripherals, then when user clicks on one of them I connect to it, read some characteristics, then navigate to second page to show some details I read. When I go back to first page and choose another peripheral, I first disconnect from connected device, then when trying to connect to the second one I chose I can connect, but when trying to read characteristic that I know is there I get "Could not find service with UUID on peripheral". 
If I do NOT navigate to second page, I can disconnect & connect without any problem.
So it looks like unmounting first page is related somehow.

tamirla, 2017-05-11:
The problem in my case was that I called BleManager.start in componentDidMount and it was called over and over again, whenever I navigated to this component.
Not sure how it's related to this error, and why it caused problems only in iOS, but once I moved it to Root component problem solved.

closed 2017-05-25 by marcosinigaglia