innoveit/react-native-ble-manager #100: BleManagerDiscoverPeripheral no getting triggered

ssozonoff, 2017-01-17:
Using your example code and running as is I see lots of BLE discoveries in the Device Log but handleDiscoverPeripheral method is never called.

ssozonoff, 2017-01-17:
Strange, seems to work now however its calling BleManager.scan() in componentDidMount() does not seem to work. No errors are reported and the scan takes place for the set duration but the BleManagerDiscoverPeripheral does not seem to get fired

marcosinigaglia, 2017-01-17:
Try to bind handleDiscoverPeripheral in the constructor

ssozonoff, 2017-01-18:
No luck :-(

marcosinigaglia, 2017-01-19:
Maybe the module is not ready in componentDidMount, try to add some delay.

ssozonoff, 2017-01-19:
Adding the delay works but it feels like a hack to me.

Thanks

marcosinigaglia, 2017-01-20:
Can't you init the module before?

closed 2017-01-30 by marcosinigaglia