innoveit/react-native-ble-manager #99: Android could not read BLE service info

wangyixian, 2017-01-17:
First, startNotification on the notification service.
BleManager.startNotification(peripheralId, '0000fff0-0000-1000-8000-00805f9b34fb', '0000fff2-0000-1000-8000-00805f9b34fb')

Second, read the notification service.
BleManager.read(peripheralId, '0000fff0-0000-1000-8000-00805f9b34fb', '0000fff2-0000-1000-8000-00805f9b34fb')

Then, write the search command into the write service. This step is return success.
 let data = base64.fromByteArray(Utils.str2Bytes('000000'));
BleManager.write(peripheralId, '', '0000fff2-0000-1000-8000-00805f9b34fb')', '0000fff1-0000-1000-8000-00805f9b34fb', data)

Could not read service info by android, but ios is ok.
Android version is 5.1

marcosinigaglia, 2017-01-17:
Hi, you don't need to use notification if you read the value directly.

wangyixian, 2017-01-17:
Thank you.
But the BLE device is designed as follow:
Write '000000' string into the write service fff1, 
then get the device state by read the notification service fff2.

Could you give me a sample about read function.

marcosinigaglia, 2017-01-17:
You have to handle the BleManagerDidUpdateValueForCharacteristic event to read the values from the service in notification.

wangyixian, 2017-01-17:
Thank you very much.
I resolved this problem by handle the BleManagerDidUpdateValueForCharacteristic event.

closed 2017-01-17 by wangyixian

hbenkhay, 2017-04-12:
I want to use notifications , I need an example to use BleManagerDidUpdateValueForCharacteristic,
does this event return an object ? can you help me please