innoveit/react-native-ble-manager #101: BleManagerConnectPeripheral

ssozonoff, 2017-01-19:
Can you confirm that BleManagerConnectPeripheral event is being fired before the Service Discovery has been completed ? Basically I have attached to the the BleManagerConnectPeripheral connection event in order to configure my peripheral but I get an error "Could not find service with UUID xxxxx on peripheral with UUID yyyy" when I try and write to it.

IMHO it would be better to receive the BleManagerConnectPeripheral once the discovery has been completed. I realise I can launch the discovery manually but since it seem to be triggered as part of the connect why do it twice ...

Thanks

marcosinigaglia, 2017-01-19:
Hi, yes the event is fired just after the connection. The success block of the promise instead wait the services discovery.

closed 2017-01-19 by marcosinigaglia

ssozonoff, 2017-01-19:
So is their en event to notify the completion of the service discovery ? How do I know when its done ? 

Thanks

marcosinigaglia, 2017-01-20:
```
BleManager.connect('XXXXXXXX-XXXX-XXXX-XXXX-XXXXXXXXXXXX')
  .then((peripheralInfo) => {
    // Service disccovery completed    
  })
  .catch((error) => {
  });
```

ssozonoff, 2017-01-20:
I understand. Now imagine I have two codepaths where a BleManager.connect could happen and I forget to add the call to init in the second one.... The idea of hooking the connect event would have guaranteed the init happen in all cases. Its no big deal I can work around it.

marcosinigaglia, 2017-01-20:
You can call the init method one time.