innoveit/react-native-ble-manager #103: Android .write results in "Write failed"

neogermi, 2017-01-23:
The following code is working fine in iOS but results in "Write failed" under Android.

iOS: 10+
Android: v23 (Samsung SM-G920F - 6.0.1)
ReactNative: 0.35.0
react-native-ble-manager: #e87659d77cd94b4d7ff668ed36ad872a50324392 (2.4.3)

```
    BleManager
    .startNotification(
      this.props.deviceId,
      "<serviceUUID>",
      "<characteristicUUIDToListenTo>",
    )
    .then(() => {
      let val = "BA=="; // should be the base64 encoded value of the integer 4
      BleManager
      .write(
        this.props.deviceId,
        "<serviceUUID>",
        "<characteristicUUIDToWriteTo>",
        value
      )
      .catch((error) => {
        console.warn("write", error); // <- this is writing "Write failed"
      });
```

What am I doing wrong here? :-(

neogermi, 2017-01-24:
I close this issue. I resolved it by surrounding a `setTimeout(..., 400);`. Apparently, Android needs some time to sort things out before writing.

closed 2017-01-24 by neogermi

pumpkin123, 2019-09-06:
I faced same problem.Thanks for solution.