innoveit/react-native-ble-manager #104: Nothing appears when trying to receive >20 bytes of data

Jami159, 2017-01-27:
I call startNotification and am successfully able to get the notifications started. Then I use the BleManagerDidUpdateValueForCharacteristic listener to get the data transmitted from my TI-CC2650. If I console.log the args (peripheral, characteristic, value), I can see that I am able to get data up to 20 bytes. But If I program my TI-CC2650 to send data >20 bytes, I get nothing when I console.log (not even any errors). I'm guessing that since the default mtu size is 20 bytes, I can't just receive anything greater than that. But I know that on android there's a requestMtu(int mtu) function that can be called to specify a larger mtu size (I'm not sure about iOS). Is there some way to implement the same functionality in this react-native-ble-manager library? (I'm very new to react-native/ble and I would really appreciate the help).

marcosinigaglia, 2017-01-30:
Hi ,the best way is to create and handle a queue for >20bytes data.

Jami159, 2017-01-31:
Sorry if this seems like a dumb question, but how exactly would I do that? My TI-CC2650 is sending data from two characteristics (fff1 and fff4). fff4 always sends a single byte of data, and I can specify how much data to fff1 should broadcast. [This is what I get when fff1 sends 20 bytes of data](http://imgur.com/VOOK99w) and [this is what I get when fff1 sends >20 bytes of data](http://imgur.com/XmvID5e). As you can see, there is nothing at all for the fff1 characteristic in the second picture (>20 bytes of data). [This](http://imgur.com/vLDZFRh) and [this](http://imgur.com/B4mQtJU) is my current code (I just modified the example code). I'd appreciate any help that you can provide. Thank you.

marcosinigaglia, 2017-01-31:
If the data is more than 20bytes you can split the message and join then in the app, you can send for example the message lenght in the first part.

closed 2017-02-01 by Jami159

adityaranjan006, 2025-02-04:
FOR IOS?