innoveit/react-native-ble-manager #106: BleManagerDidUpdateState only triggers on checkState()

JonasWho, 2017-01-30:
I do not get any state changes through the BleManagerDidUpdateState listener unless I call BleManager.checkState(). 

Is this intentional? 

I am on Android.

marcosinigaglia, 2017-01-30:
Hi, the purpose of checkState method is to force to read the current state, for example at the application start, then the event is fired when the state change.

JonasWho, 2017-01-30:
@marcosinigaglia ok, but my question is, can I expect BleManagerDidUpdateState to update when the user switches Bluetooth on and off? Or do I have to call checkState to get the state?

marcosinigaglia, 2017-01-30:
When the user change the bluetooth state the event si fired automatically

JonasWho, 2017-01-30:
@marcosinigaglia yeah, thats what I would expect. But it is not happening. Not on my Android device.

marcosinigaglia, 2017-01-30:
Which version of RN/Android do you use?

JonasWho, 2017-01-30:
RN 0.38.1, react-native-ble-manager 2.4.3 and Android 5.1.1

marcosinigaglia, 2017-01-30:
Sorry, I don't have a device to simulate your problem, try to check the permission in AndroidManifest.xml

JonasWho, 2017-01-30:
I just tested it on an Android running 6.0.1. Same problem. 

I already had `<uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>` in my manifest.

marcosinigaglia, 2017-01-30:
You have to also add `<uses-permission android:name="android.permission.BLUETOOTH"/>`

JonasWho, 2017-01-30:
Sorry, yes I already had that one as well, forgot to mention.

marcosinigaglia, 2017-01-30:
Try also `android.permission.BLUETOOTH_ADMIN`

JonasWho, 2017-01-30:
no luck with that one either :(

JonasWho, 2017-01-31:
DUH! I had not called .start() :( 
My bad, sorry for wasting your time @marcosinigaglia. The documentation could be a bit sharper IMO, but thanks for trying to get me going :)

closed 2017-01-31 by JonasWho

marcosinigaglia, 2017-01-31:
NP

KayhanB, 2018-11-08:
Is there any example code  about checkState() ?
This is my code and not fired 

import BluetoothSerial from 'react-native-ble-manager'
const bleManagerEmitter = new NativeEventEmitter(BluetoothSerial);

  componentWillMount() {
    async function REQUEST_ACCESS_COARSE_LOCATION() {
      try {
        const granted = await PermissionsAndroid.request(
          PermissionsAndroid.PERMISSIONS.ACCESS_COARSE_LOCATION,
        )
        if (granted === PermissionsAndroid.RESULTS.GRANTED) {
          console.log("You can use the bluetooth")
          BluetoothSerial.start({ showAlert: false })
            .then(() => {
              console.log('Module initialized');
              BluetoothSerial.checkState()

            });
        } else {
          console.log("Bluetooth permission denied")
        }
      } catch (err) {
        console.warn(err)
      }
    }

    REQUEST_ACCESS_COARSE_LOCATION()

    bleManagerEmitter.addListener(
      'BleManagerDidUpdateState ',
      (res) => {
        console.log(res)
      }
    )
  }

KayhanB, 2018-11-08:
Example Code  Usage checkState()

import BluetoothSerial from 'react-native-ble-manager'
const bleManagerEmitter = new NativeEventEmitter(BluetoothSerial)

BluetoothSerial.start({ showAlert: false })
            .then(() => {
              BluetoothSerial.checkState()
              console.log('Module initialized')

            });

bleManagerEmitter.addListener(
      'BleManagerDidUpdateState',
      (res) => {
        console.log(res)
      }
    )