innoveit/react-native-ble-manager #110: Unable to stop scan?

maqen, 2017-02-06:
Should it be possible to stop an ongoing scan?

I try to stop the scan upon first discovery of a peripheral but I only get the stop notification after the posted time to `startScan`

```js
_handleDiscoverPeripheral(peripheral) {
    console.log('Found peripheral', peripheral);
    BLEManager.stopScan().then(() => {
        console.log('Stopped scan');
    });
}
```

maqen, 2017-02-06:
Okay now I see the problem I get the Stopped scan but it don't seem to invalidate the timer? Get the second stopped printed out after 5 seconds?

```js
NativeAppEventEmitter.addListener('BleManagerStopScan', this._handleBLEStopScan);
_handleBLEStopScan() {
    console.log('Stopped scanning (timer)');
    this.setState({scanning: false});
}
BleManager.scan([], 5, false);
```

Do it have to be called on the main thread?

maqen, 2017-02-06:
Running version 2.4.3 see this has been solved in later versions (3.0.1), sorry for the inconvenience

closed 2017-02-06 by maqen

henninghall, 2017-02-11:
It seems like this issue still exists in android