innoveit/react-native-ble-manager #111: isPeripheralConnected will crash

wangyixian, 2017-02-07:
I used the isPeripheralConnected function as example，but react native pop up a red view。
Maybe the error was occurred from getConnectedPeripherals.

The view content as follow: 
encode
Base64.java:494

encodeToString
Base64.java:456

byteArrayToJSON
Peripheral.java:196

asJSONObject
Peripheral.java:110

getConnectedPeripherals
BleManager.java 524

marcosinigaglia, 2017-02-07:
Hi, I need more information.

wangyixian, 2017-02-08:
Sorry, I made this mistake myself.
I solved this problem. 
Thank you.

wangyixian, 2017-02-09:
The problem was reoccurred. 
BleManager.isPeripheralConnected(peripheralId, ["0000fff0-0000-1000-8000-00805f9b34fb"])
            .then((isConnected) => {
                if (isConnected) {...

was crash in android, version is 4.4 and 5.0.
But successful in ios.

marcosinigaglia, 2017-02-13:
Is it a nullpoint exeption?

closed 2017-02-14 by marcosinigaglia

wangyixian, 2017-02-23:
yes, the message is 'Attempt to get length of null array'.

marcosinigaglia, 2017-02-23:
I add a check in the last release.