innoveit/react-native-ble-manager #113: Scanning wont stop in Android Lollipop+ ?

henninghall, 2017-02-11:
I might have misunderstood something but i can't figure out why a new ScanCallback is created in stopScan21.

```
private void stopScan21(Callback callback) {

		final ScanCallback mScanCallback = new ScanCallback() {
			@Override
			public void onScanResult(final int callbackType, final ScanResult result) {
			}

		};

		getBluetoothAdapter().getBluetoothLeScanner().stopScan(mScanCallback);
		callback.invoke();
}
```

This is how stopScan function looks like in [BluetoothLeScanner](https://developer.android.com/reference/android/bluetooth/le/BluetoothLeScanner.html):

```
 public void stopScan(ScanCallback callback) {
        BluetoothLeUtils.checkAdapterStateOn(mBluetoothAdapter);
        synchronized (mLeScanClients) {
            BleScanCallbackWrapper wrapper = mLeScanClients.remove(callback);
            if (wrapper == null) {
                if (DBG) Log.d(TAG, "could not find callback wrapper");
                return;
            }
            wrapper.stopLeScan();
        }
    }

```

In other words, to stop a scanning, **the same ScanCallback** object must be sent into stopScan. Since we are creating a new mScanCallback object the previous started scan wont be stopped.  

Am I missing something here? I might be able to fix a pull request for this if needed.

marcosinigaglia, 2017-02-13:
Hi @henninghall , I think you're right, if you have a fix ready please make a pull request.
Thanks.

marcosinigaglia, 2017-02-15:
Hi @henninghall, can you test the last release? Thanks

closed 2017-02-21 by marcosinigaglia