innoveit/react-native-ble-manager #115: does it support rnpm

jjhesk, 2017-02-15:
when do you support `rnpm` scripting so that people dont have to do all the manual installations..

jjhesk, 2017-02-15:
i wanna help.. but im a newbie to iOS and still learning how to play with the xcode ide and need to get the materials learned up for rnpm.

marcosinigaglia, 2017-02-15:
Hi, rnpm is discontinued, can you test if `react-native link` works with this library?

jjhesk, 2017-02-23:
yes. i noticed that

whitedogg13, 2017-04-22:
Hi, just adapting this library to my next project and really appreciate for your work!

I have tested the tresult for using `react-native link` to link native modules for both platforms:
* iOS: work properly
* Android: still need to update `build.gradle` about `minSdkVersion`

I haven't found an easy way to directly changing `minSkdVersion`, but I think using `react-native link` is still much easier than manual linking.

I also submit a PR to adjust the README about this. https://github.com/innoveit/react-native-ble-manager/pull/135

closed 2017-04-26 by marcosinigaglia