innoveit/react-native-ble-manager #117: UI hangs up while receiving data from peripheral

namankumar, 2017-03-03:
Hi there, I've followed the instructions as you said. I can successfully receive packets from peripheral to my phones (both iOS and Android) but the phone UI is unresponsive while the data transfer is ongoing. My peripheral is sends data to the phone for 45 minutes over 24 hours, most of it in the background.  My initialization code is the same as yours except for one call:

NativeAppEventEmitter.addListener('BleManagerDidUpdateValueForCharacteristic', this.testfunc);

I use this to get and store the packets.

How can I ensure the UI is responsive when the data is being sent?

Thanks!

marcosinigaglia, 2017-03-06:
Hi, I hadn't any UI problem in my experience. How frequently is the data sending?
I don't think the module is working in background.

closed 2017-03-14 by marcosinigaglia

namankumar, 2017-03-14:
Hi, what was happening was the screen was being re-rendered with the receipt of every single packet. I made it so now the data is stored upon being received, which prevents the re-rendering. 

Everything is good now.

Thanks.

marcosinigaglia, 2017-03-15:
Ok, thanks for the reply.