innoveit/react-native-ble-manager #120: BleManager.checkState() invoke BleManagerDidUpdateState every time on app launch on IOS

shubhapp, 2017-03-14:
I just invoke BleManager.checkState() once after that when I launch APP it invoke BleManagerDidUpdateState.

marcosinigaglia, 2017-03-14:
Hi, I don't understand the problem.

shubhapp, 2017-03-14:
[in case of ios] If I called BleManager.checkState() once and after that I killed my application then on relaunch of the app, BleManagerDidUpdateState listener will be invoked (without calling BleManager.checkState() again)

marcosinigaglia, 2017-03-15:
BleManager.checkState only force to fire the BleManagerDidUpdateState event to read the current state. BleManagerDidUpdateState is fire also automatically when the state change in the app.

shubhapp, 2017-03-17:
BleManagerDidUpdateState is fire also automatically when the state change in the app. in this case you mean if Bluetooth state changes. correct ?

marcosinigaglia, 2017-03-17:
Yes.