innoveit/react-native-ble-manager #122: Android Scan not working?

WillDent, 2017-03-24:
I have this running fine on iOS and able to build the example and compile with no issues to Android. However the farthest it gets is attempting to scan over and over.

''  handleScan() {
        BleManager.scan([], 30, true)
            .then((results) => {console.log('Scanning...'); });
    }

This function is called over and over and never has any results.  I added a catch to the promise and never any error messages either.  Very odd.

This is using Android 6.0 and building directly to a device (Samsung Galaxy S7)

marcosinigaglia, 2017-03-24:
Hi, you have to see in the device log and check for some problem.

WillDent, 2017-03-29:
My apologies this is working fine.  I am not sure why it is working now but I just cycled the BLE on the device and seem to unstick this.  @marcosinigaglia

closed 2017-03-29 by WillDent

Jayashree-panda, 2020-10-11:
@WillDent Hi, I am also facing a similar issue. Can you please let me know the changes you made that started the scan option working for your device?