innoveit/react-native-ble-manager #123: initWithBase64EncodedString

develand, 2017-03-27:
I am sending 22 bytes to the device that have been Base64 encoded as suggested in the documentation.  I am trying to understand why initWithBase64EncodedString is called within the write method (on iOS).  My device is expecting my 22 Base64 bytes to be transmitted, but it looks like this is translating into 36 bytes which are then written to the device.  I apologize in advance if this is based on my lack of understanding but I have scoured the code and documentation and my device responds correctly if I used Nordic's tool to send the bytes, but not if the write method is called.

marcosinigaglia, 2017-03-29:
Hi, I use the Base64 encoding because is the easiest way I finded to use the same method in iOS and Android. The native module receive the string encoded and convert back in the byte array that is send to the bluetooth device. I don't understand your problem sorry.

develand, 2017-03-29:
This issue can be closed.  I found the problem on my side (finally!).  The problem was my failure to make sure the string was properly encoded in base64 format.

closed 2017-03-29 by develand