innoveit/react-native-ble-manager #125: Receive bluetooth Data speed?

zhanguangao, 2017-04-01:
Android 5.0 and Android 6.0 receive bluetooth data faster than Android 7.0 more than twice as fast, IOS10  is also slow.Hou can I do to improve receive bluetooth data speed for high version of the system.

zhanguangao, 2017-04-01:
Android 5.0 and Android 6.0 log:
04-01 16:13:32.030  ReactNativeJS: 'Write success: ', 'FEFD810EFCFB'
04-01 16:13:32.105  ReactNativeJS: 'BluetoothUpdateValue', 'FEFDA18E000000000000005045502D4843303031'
04-01 16:13:32.137  ReactNativeJS: 'BluetoothUpdateValue', '04803B110000000000030201450C2E02FCFB'

Android 7.0 log:
04-01 16:44:11.278  ReactNativeJS: 'Write success: ', 'FEFD810EFCFB'
04-01 16:44:11.465  ReactNativeJS: 'BluetoothUpdateValue', 'FEFDA18E000000000000005045502D4843303031'
04-01 16:44:11.599 ReactNativeJS: 'BluetoothUpdateValue', '04803B110000000000030201450C2E02FCFB'

marcosinigaglia, 2017-04-03:
Hi, I don't think you can improve the speed rate, which write mwthod do you use?

zhanguangao, 2017-04-03:
writeWithoutResponse

marcosinigaglia, 2017-04-04:
Is the fastest method, try to search in the android documentation for hints.

zhanguangao, 2017-04-04:
I see. Thank you!

closed 2017-04-12 by zhanguangao

McDreamy, 2021-02-19:
How to get the data from writeWithoutResponse on another device?