innoveit/react-native-ble-manager #126: Device is not re-discovered after a disconnect

vldgeorgiev, 2017-04-05:
There is one problem with the connect/disconnect functionality.
When a device is discovered it is added to the "peripherals" list in BleManager.java and a notification is sent. If a device exists in the list, there is no notification to the app.
The issue is that the device is not removed from there after it is disconnected (either from the device or from the app). So the app doesn't receive further discovery notifications about this device and it is not possible to reconnect to it without restarting the scan, which purges the disconnected ones from the list.

I wonder if this is by design and if it is how do you work around it?
For my usage it doesn't work because I want to keep the scanning without restarting it, but also have to reconnect to devices several times. To do this I modified the BleManager to clean up disconnected devices, and to re-send notifications about them.

If this functionality is not intended I can make a pull request with the changes. I checked the Android code only, but the iOS one seems similar.

marcosinigaglia, 2017-04-06:
Hi, when you disconnect from a peripheral you can reconnect without scan again so is not an issue. The scan is expensive so if you have arleady finded a peripheral you can directlty connect and disconnect for many times.

vldgeorgiev, 2017-04-06:
Sorry, I didn't describe it very well.
Yes, if the device is on, I can reconnect to it at whenever I want. In my case the device gets turned off and I get a notification about it, or I disconnect explicitly and then it turns off. And after it is turned on again and is re-discovered I want to get a new notification about it without restarting the scan. This is the part that can not happen in the current implementation.

marcosinigaglia, 2017-04-06:
I never tried but I don't think you have a discover event if you don't scan again.

vldgeorgiev, 2017-04-06:
Yes, you don't get one. The device is discovered internally in the manager, but the event is not passed to the app.
To me it sounds more logical to pass it to the app and let it decide what to do, but you are the authors, so it is up to you.

marcosinigaglia, 2017-04-06:
I think the best way is to create a specific option to the `start` method to disable the check.

vldgeorgiev, 2017-04-06:
Sure. I will test it some more in a couple of days and make a PR for the changes.
It will probably require an extra public property in the Peripheral class, in order to keep the device in the list, but still have the discover event fire again once.

closed 2017-04-06 by vldgeorgiev

marcosinigaglia, 2017-04-06:
Ok, thanks.

stephenjen, 2017-08-29:
Was this implemented for iOS or just Android?