innoveit/react-native-ble-manager #127: 'connect' method is slow

vldgeorgiev, 2017-04-09:
Hi again

The BleManager.connect method seems to take a lot of time to establish the connection and report it back. Usually about 7-8 seconds, even after the scan is started and the device has been discovered.

This would be an issue for users, because after they turn the BLE device on, it would take like 10 seconds for it to be discovered and connected, which is quite long.
I am comparing with another BLE testing app where the same device takes 2-3 seconds for discovery and connection. On an iPhone the results are similar.

Any idea what could be the reason for the extra 6-7 seconds?

The code in use is something like:
```
NativeAppEventEmitter.addListener('BleManagerDiscoverPeripheral', this.handleDiscoverPeripheral);
...
BleManager.scan([], 0).then((results) => { console.log('Scanning...'); });
...
handleDiscoverPeripheral(data) {
  console.log('Connecting');
  BleManager.connect(data.id)
  .then(() => {
    console.log('Connected', data.id);
  })
};
```
Below is an example of the log data:
The discovery itself takes up to 1-2 seconds, which I suppose is OK.
However, the delay between the "Connecting' and 'Connected' is very long. There are about 5sec before the onClientConnectionState event is received, and about 2 more till the Promise is fulfilled and the app is notified about the connection.

```
04-08 11:51:45.744: I/logs(11219): DiscoverPeripheral: AAA
04-08 11:51:45.757: I/ReactNativeJS(11219): 'Connecting', '18:7A:93:03:1F:C0'
04-08 11:51:45.759: D/logs(11219): Connect to: 18:7A:93:03:1F:C0
04-08 11:51:45.760: D/BluetoothGatt(11219): connect() - device: 18:7A:93:03:1F:C0, auto: false
04-08 11:51:45.760: D/BluetoothGatt(11219): registerApp()
04-08 11:51:45.760: D/BluetoothGatt(11219): registerApp() - UUID=1a3729e8-ac53-404b-8381-309ffccadbd7
04-08 11:51:45.763: D/BluetoothGatt(11219): onClientRegistered() - status=0 clientIf=6

04-08 11:51:50.897: D/BluetoothGatt(11219): onClientConnectionState() - status=0 clientIf=6 device=18:7A:93:03:1F:C0
04-08 11:51:50.897: D/logs(11219): onConnectionStateChange from 0 to 2 on peripheral:18:7A:93:03:1F:C0
04-08 11:51:50.897: D/BluetoothGatt(11219): discoverServices() - device: 18:7A:93:03:1F:C0
04-08 11:51:50.902: D/logs(11219): Peripheral event (eventName):18:7A:93:03:1F:C0
04-08 11:51:52.297: D/BluetoothGatt(11219): onSearchComplete() = Device=18:7A:93:03:1F:C0 Status=0
04-08 11:51:52.327: I/ReactNativeJS(11219): 'Connected', '18:7A:93:03:1F:C0'

```

marcosinigaglia, 2017-04-10:
Hi, the callback in the `connect` method is call after the services and characteristics discover so you can use the `BleManagerConnectPeripheral` event to check the connection. It is still strange that need so much to read the services.

vldgeorgiev, 2017-04-10:
Yes, I also tested the BleManagerConnectPeripheral event.
It is somewhat faster and is received about 1 sec before the connect callback, but it is still a long delay. I wonder what is done during this time. The underlying Java code is quite short and I don't see a reason for the delay.
So you haven't had such experience with the connection delays?

marcosinigaglia, 2017-04-10:
No, have you tried to use your app disabling the debug or with a release build?

vldgeorgiev, 2017-04-10:
Yes. The times in the release build are the same. Could be some specific
issue with this device and this library, because with another app it
connects quickly.
I don't have another ble device to test with.

marcosinigaglia, 2017-04-10:
Do you have an Android device? Version?

marcosinigaglia, 2017-04-10:
I saw now log and the delays are not caused by RN. Do you have the BT device documentation?

vldgeorgiev, 2017-04-10:
Yes, it is not caused directly by the RN module. It seems to come from connectGatt, but there must be a way around it, which other apps seem to use. There is some info here http://stackoverflow.com/questions/22214254/android-ble-connect-slowly, but your Java code already uses this recommendation.

For the BLE device I have only brief info about the read/write commands. Other than that, it exposes about 5 services with a couple of specifications each. Do you think that discovering the services and spec could be slow? They are not that much.

marcosinigaglia, 2017-04-10:
I don't think the problem is in the discovering, try to force the use of the LollipopScanManager in the BleManager constructor and see if change something.

vldgeorgiev, 2017-04-10:
Maybe you meant the Legacy scanner? I tried forcing it instead of the Lollipop one and it improved the speeds on the Android 6.0. Now it takes 1-4 seconds, which while not the best, is much better.
I'll use it like that for a while and see how it goes.

marcosinigaglia, 2017-04-10:
Yes sorry, I mean the legacy. I'm already planning to add a config flag for
that.

marcosinigaglia, 2017-04-18:
I added the flag.

closed 2017-04-18 by marcosinigaglia