innoveit/react-native-ble-manager #128: "TypeError: bleManager.enableBluetooth is not a function" when debugging JS remotely

itsbigk, 2017-04-13:
I'm getting this error when running BleManager.enableBluetooth. It seems to only happen when I enable remote JS debugging in my app.

Is there something specific I need to change to get this to work? Any help would be greatly appreciated.

marcosinigaglia, 2017-04-18:
Hi, what do you mean "remotely"? Is the only method that cause problem?

itsbigk, 2017-04-19:
@marcosinigaglia Thanks for the reply. I was actually able to get it working. Not entirely sure how. Just a lot of trial and error. Still not sure exactly why this happens but it does work eventually.

What I meant by "remotely" is using the remote js debugging feature of react native. This allows the react native code to be debugged using the chrome web tools. More information about this can be found at https://facebook.github.io/react-native/docs/debugging.html#chrome-developer-tools

closed 2017-04-19 by itsbigk

marcosinigaglia, 2017-04-20:
Hi, I always use the debug in my projects with no problems, check if is all ok in the installation.